sparkfun/Arduino_Apollo3 #1: Feature Request: Generic QWIIC/Wire.h interface

karma0, 2019-04-20:
### Subject of the issue
I2C and SPI interfaces as a drop-in replacement for the Wire.h Arduino library with the QWIIC ecosystem would allow users to interface with relays, sensors, displays, and plenty of other external devices.

### Expected behaviour
I should be able to write code to a device's implementation as described.  As many devices use the Wire.h library, it should be as easy as porting this to the Apollo3.

### Actual behaviour
No such library exists, requiring users to have to research the Apollo3 SPI/I2C interfaces and create their own custom drivers for selected hardware that typically already has an implementation in existence.

oclyke, 2019-04-22:
Yes! Happy to work on this one. I2C and SPI (as you seem to know) both use the IO Master peripherals on the Apollo3. Timeline is probably shortly after we get a working "Serial" interface.

oclyke, 2019-05-16:
@karma0 I just pushed preliminary I2C support. For now it only supports master mode. Will you give it a try and tell me what you think? 

The example to start out with is ```libraries/CoreTesting/examples/example5_Wire```

For now you'll want to use the Edge board variant -- it has Wire (on the Qwiic connector) and Wire1 (attached to the onboard Accelerometer) defined. Right now the other variants may not have any TwoWire objects pre-defined (but you can always declare your own like ```TwoWire MyWireObj(IOMasterInstance)``` where IOMasterInstance is an integer 0-5 and it determines which pins the object is associated with.

SPI support is up next on the list!

oclyke, 2019-05-20:
And now I've pushed preliminary SPI support. 

Both of these features are bare-bones at the moment. For example I2C is master-only and SPI does not handle any interrupt functions. Since these are part of the Arduino IDE we expect to get to them eventually, however at the moment there are other more pressing developments to be made. 

If incompatibilities/missing features are found let's make an issue and address them as they arise. Thanks for this issue BTW!

closed 2019-05-20 by oclyke

karma0, 2019-05-20:
Thanks @oclyke!  I haven't had the opportunity to try these out, but I can always just open another issue if I can't get it to work.