sparkfun/Arduino_Apollo3 #14: Build out support for PlatformIO

nseidle, 2019-06-26:
I've never done it but if someone has the experience getting an platform into platformio I would love to work with them. Please let me know.

nigelb, 2019-12-01:
Hi, I wouldn't say I am experienced at development for PlatformIO but I have been playing with this:  [https://github.com/nigelb/platform-apollo3blue](https://github.com/nigelb/platform-apollo3blue).  It has not received a lot of testing because I ran into this issue: sparkfun/Arduino_Apollo3#94

jerabaul29, 2020-12-05:
Just curious of what the status is here? :)

nigelb, 2021-06-11:
Hi, I and have recently fixed a lot of bugs (in [platform-apollo3blue](/nigelb/platform-apollo3blue)) and got thing working pretty well so that the master can now build against both the v1.x.x and v2.x.x series of Arduino_Apollo3. Now that I have identified (sparkfun/Apollo3_Uploader_SVL#4) and found a work around for (sparkfun/Apollo3_Uploader_SVL#6) my last remaining issue with uploading a firmware from Linux I think it is getting to the point where, soon, I can publish the apollo3blue platform to the PlatformIO repository.

For this to happen we need to sort out the following in Arduino_Apollo3:

1. Version bump the [ `package.json`](https://github.com/sparkfun/Arduino_Apollo3/blob/9f309a09712637a58e334981d000e1ff78aa575e/package.json#L4) of the 1.x.x series.
2. Add a `package.json` for the v2.x.x series (I am happy to submit a pull request):
```json
{
    "name": "framework-arduinoapollo3",
    "description": "An mbed-os enabled Arduino core for Ambiq Apollo3 based boards",
    "version": "2.1.2",
    "url": "https://github.com/sparkfun/Arduino_Apollo3"
}
```
3. With the release of PlatformIO 5 they removed the ability to point to other [repositories](https://docs.platformio.org/en/latest/core/migration.html#packagerepositories) for packages. What this means is that both versions of Arduino_Apollo3 will need to be published to PlatformIO repositories.

jerabaul29, 2021-11-17:
Just being curious: any progresses forward on this? I can personally install / tweak by hand things, but working with some students / other users who have difficulties getting the platformio package going and switching between core v1 and core v2 there.