sparkfun/Arduino_Apollo3 #16: Add support for Servo

nseidle, 2019-06-26:
Add support for the Arduino library Servo. Owen's already got the analogWrite() function examples that show how to do this. We just need to build out the other parts of the library API.

nseidle, 2019-06-26:
Weird short blip on the pin when using servoWrite(). No load (servo) is attached.

[image]

[image]

The blips are 15ms apart so it looks like an analog write issue.

oclyke, 2019-07-03:
@nseidle caught the moment of inversion on a logic analyzer.
[image]

You can see that the timing between servo frames was 21.9 ms consistently. You can also see regular 15 ms spacing between short blips toward GND after the inversion. You will find that these two frequencies meet perfectly at the E1 marker - GND zero. 

The test code was waiting 15 ms between each call to servoWrite(). When that delay collided with the asynchronous timer comparison bad things happened (* insert some explanation for what happened in the bowels of the silicon * ) and the signal was inverted.

Commit 82089ed tries to solve this in a stateless manner by ensuring that the timer value is less than the smaller of the two compare values. Initial testing looks promising, however it's still possible that the servoWrite() function tries to change the register values right when ctimer_val == cmpr0_val. I don't know what will happen in that case. 

This solution is nice because it avoids the need for an 'is initialized' array for analog output pins.

oclyke, 2019-08-05:
Closing until we hear more...

closed 2019-08-05 by oclyke