sparkfun/Arduino_Apollo3 #21: Upload issue SparkFun Edge

metanav, 2019-07-04:
### Subject of the issue
I am not able to upload code to SparkFun Edge using Arduino IDE. It always keeps waiting at "sending hello". It seems the bootloader is not responding. 

### Your workbench
* What platform are you using?
MacOS Mojave
* What version of the device are you using? Is there a firmware version?
I am using pre-oder SparkFun Edge directly from SparkFun.
* How is the device wired to your platform?
Using SparkFun Serial Basic Breakout - CH340C and USB-C
* Are there any additional details that may help us help you?
Using Ambiq SDK toolchain there is no such issue.

nseidle, 2019-07-08:
Follow [these instructions](https://learn.sparkfun.com/tutorials/designing-with-the-sparkfun-artemis#troubleshooting) to install the faster/better variable bootloader.

Explanation: Edges came pre-programmed with the Ambiq secure bootloader. The SparkFun Arduino core defaults to the SparkFun Variable Bootloader (SVL). We recommend upgrading to the SVL or you can select the  Ambiq Secure bootloader and try again.

[image]

oclyke, 2019-07-08:
One more comment... The pre-order version of the Edge board should expect to be boot-loaded at 921600 baud (we chose this for decreased bootload time), however in the Arduino core the Ambiq Secure Bootloader (SBL) is currently fixed at 115200 baud (which helps for sure-fire compatibility across operating systems because of the USB-serial driver discrepancies). So right now simply selecting 'Ambiq Secure Bootloader' won't work out of the box. 

There are at least two possible fixes:
1. Add an SBL baud rate property in the boards.txt file, and create two different Edge boards (one for TensorFlow summit and one for preorder) 
2. Add a menu property to allow selecting the SBL baud rate. 

@metanav Do you have a preference between these two options? @nseidle how about you? We can work this into the next release of the core. 

@metanav As a temporary workaround... you can edit 'platform.txt' in your installation to use 921600 baud for the SBL. Change:
tools.ambiq_bin2board.args.macosx=--bin "{build.path}/{build.project_name}.bin" --load-address-blob 0x20000 --magic-num 0xCB -o {build.path}/{build.project_name} --version 0x0 --load-address-wired 0xC000 -i 6 --options 0x1 -b 115200 -port "{serial.port}" -r 2 {upload.verbose}

to 

tools.ambiq_bin2board.args.macosx=--bin "{build.path}/{build.project_name}.bin" --load-address-blob 0x20000 --magic-num 0xCB -o {build.path}/{build.project_name} --version 0x0 --load-address-wired 0xC000 -i 6 --options 0x1 -b 921600 -port "{serial.port}" -r 2 {upload.verbose}

oclyke, 2019-07-08:
FYI re-opened as a reminder to fix this for the next release

metanav, 2019-07-08:
Thanks @nseidle and @oclyke. It is working now after the changes suggested above. But I had to do few things to make it work. 

In the directory /Arduino15/packages/SparkFun/hardware/apollo3/1.0.1/tools/ambiq/macosx/ambiq_bin2board:

While uploading it was complaining about missing modules am_defines and keys_info. For the workaround I created symlinks as follows:
```
am_defines.py -> ../../am_defines.py
keys_info.py -> ../../keys_info.py
```
In the am_defines.py, comment out following imports (I tried to install using pip but still not loading so I commented out):
```
#from Crypto.PublicKey import RSA
#from Crypto.Signature import PKCS1_v1_5
```

Also, I had to change this in platform.txt:
`tools.ambiq_bin2board.pgm.macosx={runtime.platform.path}/tools/ambiq/macosx/ambiq_bin_to_board/ambiq_bin_to_board
`
to:

`tools.ambiq_bin2board.pgm.macosx={runtime.platform.path}/tools/ambiq/macosx/ambiq_bin2board/ambiq_bin2board
`
And set  execute permission:
`chmod +x ambiq_bin2board
`

I hope these issues will be fixed in the next release.

oclyke, 2019-07-09:
Thanks @metanav, I will use your notes to improve this soon!

oclyke, 2019-07-10:
@metanav I've just released [v1.0.2](https://github.com/sparkfun/Arduino_Apollo3/releases/tag/1.0.2) to address upload concerns. I fixed the paths, and generated the binaries with "pyinstaller --onefile" to avoid making those symbolic links. On Mac and Linux I also added executable permissions, however I don't know enough to be sure that those changes will be respected when the core is installed by Arduino. 

For the Edge board I added the ability to choose the baud rate for the Ambiq Secure Bootloader, as well as the advanced option to enable the SparkFun Variable Loader.

Will you please try out the new version and let me know how it goes? Thank you.

metanav, 2019-07-10:
Thanks @oclyke, It works! Though it is still needed to set executable permission.
Also, I uploaded  Artemis Examples > Advanced > Artemis_Bootloader with Ambiq Secure Bootloader which was succeeded. But changing to SVL bootloader the code upload fails with error:
```
Waiting for command from bootloader
No announcement from Artemis bootloader
```
The Ambiq bootloader still works.

oclyke, 2019-07-10:
Okay glad that there is some progress! Seems like there are two remaining items:

1. **Executable Permissions**
 I was able to recreate the permissions error on my own hardware:
- MacOSX 10.14.5 (18F132)
- Arduino 1.8.9
- Apollo3 v1.0.2 freshly installed
- Edge board with SBL @ 115200

*Update* This should be solved now. Use the boards manager to remove your current 1.0.2 installation and then re-install it. The [JSON Boards Manager file](https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json) is updated to use the ".tar.gz" archive instead of the ".zip" archive that was used before. This should preserve the executable file permission.

1. **Artemis SVL No Announcement**
I was able to reproduce a similar error on my Mac, but not on my Windows machine. I'll look into this further.

oclyke, 2019-07-11:
We've looked into the "No Announcement" error more and discovered discrepancies between Windows and MacOSX in the Python script timing and control line handling. The bootloader is sensitive to these differences and needs to be fixed. 

This is no longer just an Edge upload issue, so we can continue the discussion here: 
https://github.com/sparkfun/SparkFun_Artemis/issues/2

closed 2019-07-11 by oclyke

oclyke, 2019-07-16:
@metanav I just merged a PR to fix this. Can you test it out by replacing the copy that Arduino installs with one directly from the Repo? I'd like to have another data point before making the Arduino release. Thanks!

metanav, 2019-07-17:
@oclyke After latest changes it is still not uploading:

```
Sketch uses 7356 bytes (0%) of program storage space. Maximum is 960000 bytes.
/Users/mylogin/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.3/tools/artemis/macosx/artemis_svl /dev/cu.wchusbserial1420 -f /var/folders/ck/1n_m22xn15zdntmwtj9bvx400000gn/T/arduino_build_993898/Example1_Blink.ino.bin -b 921600 -v 
Connecting over serial port /dev/cu.wchusbserial1420...
Using Serial timeout: 0.05
Starting Communications at 921600 baud
Try #19. Result: 
Baud Set: lengths did not match
Failed baud set phase
/dev/cu.wchusbserial1420 is currently open. Please close any other terminal programs that may be using /dev/cu.wchusbserial1420 and try again.

```

oclyke, 2019-07-17:
@metanav Fantastic to hear! We will soon add documentation -- clearly you knew to upload the new SVL bootloader sketch with the Ambiq SBL and then switch bootloaders to SVL but we will want others to have an easier time. We will work on making 'burn bootloader' perform that same job. 

The point about Button14 is a good one - SVL relies on the board not going into the Ambiq SBL, which is activated on reset if the button is held down. All future Artemis boards by SparkFun will not include Button14, but I will try to find a good place to make mention of that fact for others.

asanjib28, 2022-04-23:
I struggled to upload a simple blink code into a Sparkfun Edge board, for the past one month, but did not succeed yet. In fact I purchased a 2nd board as well thinking that I might have damaged the first board. But the second also behaved the same. I am using a Sparkfun recommended USB-C to UART (3.3V) breakout, which uses CH340C. My computer has Windows 10 and I am have tried both Arduino IDE 1.8.19 version and Aduino IDE 2.0.0-rc6. But no success.

When I use Arduino IDE 1.8.19, my settings are:
[image]

Then when I press the upload button, it tries three times and then fails. I tried reducing baudrate. I tried pressing buttons as mentioned in the Sparkfun tutorial. Again I have also tried with not touching any buttons. I could see that Tx LED on the CH340C breakout board blinks momentarily but Rx does not blink. The edge board gets reset when the Tx LED blinks (if I do not keep the button 14 pressed on), and it does three times before declaring fail to upload: 
================================================================
Header Size =  0x80
original app_size  0x2224 ( 8740 )
load_address  0xc000 ( 49152 )
app_size  0x2224 ( 8740 )
w0 = 0xcb0022a4
Security Value  0x10
w2 =  0x10008080
addrWord =  0xc000
versionKeyWord =  0x0
child0/feature =  0xffffffff
child1 =  0xffffffff
crc =   0xf331a6c3
Writing to file  C:\Users\Sanjib\AppData\Local\Temp\arduino_build_222327/Blink_SparKFunEdge.ino_OTA_blob.bin
testing: C:\Users\Sanjib\AppData\Local\Temp\arduino_build_222327/Blink_SparKFunEdge.ino_OTA_blob.bin
Header Size =  0x60
app_size  0x22a4 ( 8868 )
Writing to file  C:\Users\Sanjib\AppData\Local\Temp\arduino_build_222327/Blink_SparKFunEdge.ino_Wired_OTA_blob.bin
Image from  0x0  to  0x22a4  will be loaded at 0x20000
Connecting over serial port COM16...
Sending Hello.
No response for command 0x00000000
Failed to respond
Fail
Sending Hello.
No response for command 0x00000000
Failed to respond
Fail
Sending Hello.
No response for command 0x00000000
Failed to respond
Fail
Tries = 3
Upload failed
=======================================================================

I am exhausted trying. Any lights please....

Regards,
Sanjib

metanav, 2022-04-25:
Sometimes USB cable can be an issue. Give it a try with another better USB cable.

asanjib28, 2022-04-30:
I was following the upload instructions mentioned on Sparkfun website, which are not accurate. I was releasing button 14 before the upload was completed, which caused this issue:
https://github.com/sparkfun/Arduino_Apollo3/issues/461#issuecomment-1107615170