sparkfun/Arduino_Apollo3 #22: BLE support

VALERE91, 2019-07-07:
Hi, I'm considering to use the Artemis platform for my next project which require BLE capabilities. Is there any effort currently done to support BLE with the Arduino core?

Thanks

oclyke, 2019-07-08:
Yes, that is one area which we are focusing on right now. In case you want to get started sooner there is already BLE support in the AmbiqSuite Hardware Abstraction Layer which can be used within an Arduino sketch. Download the [SDK](https://ambiqmicro.com/static/mcu/files/AmbiqSuite-Rel2.2.0.zip) and look into the ```mcu/hal``` directory for the source files that could be applicable.

VALERE91, 2019-07-08:
Thanks for the precision. I will check the HAL source code and maybe adapt a GATT BLE library for it.

metanav, 2019-08-23:
Any progress?

VALERE91, 2019-08-23:
Hi @metanav, no sorry with the summer and my job I did not have enough time to put on that task. I tried the HAL and finally decided to use the Ambiq SDK for now on. Harder but with good examples on BLE usage with ExactLE.

metanav, 2019-08-23:
Yes I did the same.  But I am looking forward for Arduino Core API.
Ping @oclyke

oclyke, 2019-08-23:
Working on it as we speak. I spent quite a while trying to get familiar with the ExactLE stack as well as trying to choose a good (future proof?) API to follow for the implementation. 

For the time being I have decided to try to get a bare minimum example (ble_cordio_tag) woring within the IDE. You can actually see the work in-pogress on the branch 'ble-core'. In case you're wondering it is named that way b/c it is BLE support directly in the core rather than as a library. This is not the ideal solution, IMO, because it means that the BLE files will get compiled every time the core is. Unfortunately the alternative -- to keep all BLE files in a BLE library -- is substantially harder because of the way that Arduino handles include paths for libraries. 

The current state of the branch is that code can be compiled and I expect it to function just like the SDK example but the process gets hung up somewhere. I am trying to debug it.

If you have the interest you could check it out -- I could definitely use some outside perspective on how best to structure BLE support in the long term

oclyke, 2019-08-23:
@VALERE91 @metanav 

Good news -- I was able to work through some of the trouble that I explained above. Currently the ```ble-core``` branch contains an example that sets up the Apollo3 as a BLE peripheral advertising with the name 'Tag' and these services: ```Link Loss``` ```Immediate Alert``` and ```Tx Power```. The example is awkwardly placed right in the root directory and is called ```zzz_temporary-ble-build-out```.

There is still a lot of work to be done. Namely how to allow the user to easily create their own set of services as well as callbacks to handle when those services are accessed. In the meantime check it out! You can try changing the advertised name from 'Tag' to 'Wag' for example (do that in nus_main.c)

mbr4477, 2020-03-21:
Has there been any thought of using the [Particle BLE APIs](https://docs.particle.io/reference/device-os/firmware/xenon/#bluetooth-le-ble-) as a template or starting point for the Apollo3 interfaces? I've used the Particle APIs with their Xenon board and was easily able to BLE setup services in that paradigm.

linkoep, 2020-04-01:
Just trying to wrap my head around this and get an idea of current progress.
All HAL functions required to implement BLE functionality are in a form where they can be used in an arduino sketch, as per above.
As per [sparkfun's announcement](https://www.sparkfun.com/news/3192) and [the most recent forum post](https://forum.sparkfun.com/viewtopic.php?f=169&t=50915&start=15) I could find, the end goal is to have the standardized ArduinoBLE.h interface on the Artemis, and to do so by leveraging mbed os.

1. What is the current recommendation for BLE on the artemis platform? Using HAL functions? Switching to the AmbiqSuite SDK? Does that SDK have a relatively simple BLE interface, or does it also primarily use HAL functions directly?
2. What work (very broadly) is left to do to achieve the ArduinoBLE.h support on artemis? What is the relationship between the mbed os work and the support of this interface?
3. Could (or has?) someone implement the ArduinoBLE.h interface on the artemis by using HAL calls directly, independently of the mbed os work?

Thank you for all the work you've put into the platform so far!

oclyke, 2020-04-02:
@linkoep Thanks for clearly expressing your questions - that made it easy to address them. I hope these answers will illuminate what's going on. If they don't do the trick just say so.

1. The AmbiqSuite SDK effectively *is* the HAL. So using the SDK will be the same as using HAL functions without access to any Arduino creature-comforts (Serial)

2. The work remaining involves completing a port of mbed for the Apollo3. (that work is [here](https://github.com/sparkfun/mbed-os-ambiq-apollo3/branches) and the best stable branch is [ambiq-apollo3](https://github.com/sparkfun/mbed-os-ambiq-apollo3/tree/ambiq-apollo3)) The way it relates is that the Arduino Nano 33 BLE currently implements the ArduinoBLE.h interface using mbed. If Apollo3 supports mbed the implementation should be nearly identical. Completing the port (well) requires testing which requires a lot more work and development in projects such as pyOCD, DAPLink, and FlashAlgo. Look for our forks of those as well and see the Mbed OS5 porting guide: https://os.mbed.com/docs/mbed-os/v5.15/porting/index.html

If you set up the mbed command-line interface (cli) correctly you should be able to compile supported examples like:
- [Blinky](https://github.com/ARMmbed/mbed-os-example-blinky)
- [BLE LED](https://github.com/ARMmbed/mbed-os-example-ble/tree/master/BLE_LED)

Using a command along the lines of ```mbed compile --target sparkfun_artemis_thing_plus --toolchain gcc_arm```

You would then need to flash the binary file into the Apollo3 at the correct address (it should be either 0x0000C000 or 0x00010000 - but we've seen some odd behavior at times) using either the serial bootloaders (ASB / SVL) or directly with a debug tool like a SEGGER J-Link EDU ($20 ish?)

3. Yes, you could. I haven't heard of anyone taking that route. If someone wanted to put that level of effort into the project I would suggest to them that they can / should aid in the mbed port. Better to standardize and gain additional benefits (like well-written unit tests) (Plus - the goal is to build the Arduino core on top of mbed for version 2.0.0)

Wenn0101, 2020-10-12:
ArduinoBLE Support was added to v2.0.0 of the arduino core. Closing.

closed 2020-10-12 by Wenn0101