sparkfun/Arduino_Apollo3 #26: SPI needs setClockDivider

nseidle, 2019-07-09:
Attempting to run [SparkFun BME280](https://github.com/sparkfun/SparkFun_BME280_Arduino_Library) library failed because 

    C:\Dropbox\Apps\Arduino\libraries\arduino_173658\src\SparkFunBME280.cpp: In member function 'uint8_t BME280::begin()':
    C:\Dropbox\Apps\Arduino\libraries\arduino_173658\src\SparkFunBME280.cpp:98:7: error: 'class 
    SPIClass' has no member named 'setClockDivider'
        SPI.setClockDivider(SPI_CLOCK_DIV32);

nseidle, 2019-07-09:
FWIW it's not recommended for use in new projects: https://www.arduino.cc/en/Reference/SPISetClockDivider

nseidle, 2019-07-09:
Notes to myself: we're not respecting LSB MSB settings. I believe the HAL has MSB hardcoded:

[image]

Page 321 of datasheet shows bit 23 controls MSB/LSB so we should be able to manipulate it. I know there are bit wise operators in the HAL...

[image]

nseidle, 2019-07-09:
Register access helper macros: 

[image]

nseidle, 2019-07-09:
The macro helper functions are defined in v2.2 of HAL so we don't have access to them but the AM_REGVAL is available and seems to be working.

nseidle, 2019-07-10:
Should be fixed in PR #30.

oclyke, 2019-08-05:
All this is taken care of now. Thanks!

closed 2019-08-05 by oclyke