sparkfun/Arduino_Apollo3 #36: Binding Arduino Binaries with TensorFlow Models

RicoFio, 2019-07-24:
### Subject of the issue
As I am currently working on a project with the [SparkFun Edge Dev Board ](https://www.sparkfun.com/products/15170) which uses the Ambiq Apollo 3, I'm wondering if there is a way to bind the Arduino generated binaries with a TensorFlow Lite build for the Edge?

### Your workbench
* MacOS Mojave 10.14.5 (18F132)
* SparkFun Edge Dev Board V1
* USB-C Breakout UART

oclyke, 2019-07-25:
@RicoFio I'm happy to try and help... but you probably know more about the TensorFlow stuff than I do. I don't understand what you mean by 

> bind the Arduino generated binaries with a TensorFlow Lite build 

Can you explain more about what you are trying to do?

RicoFio, 2019-07-25:
Hi @oclyke,
Thanks for coming back on this one! As you already saw in [this issue](https://github.com/sparkfun/SparkFun_Edge_BSP/issues/12#issue-471777680) I was trying to get the SPI example to work. What I am actually up to is to connect two edge devices via I2C (using the qwiic connector). That actually worked fine with the update Arduino Core in the IDE!

However, as both will be running TensorFlow code ([like here](https://codelabs.developers.google.com/codelabs/sparkfun-tensorflow/index.html#0)), I'll need to integrate the I2C in the C Code.

My attempt today was to write a C wrapper for the Arduino Core's C++ Wire Library ([this one for reference](https://github.com/sparkfun/Arduino_Apollo3/tree/master/libraries/Wire/src)). I haven't tried my implementation yet, but do you think that's the right direction? Or is there an easier way? I have been scared off a bit by the Ambiq Library for C which is used in the SPI Example you have in the Arduino_Apollo3 but if that's the easiest way, I guess I'll try to dig my way through that for the I2C.

Thanks a lot in advance for any ideas!

oclyke, 2019-07-25:
@RicoFio I think I understand better now! 

As a note about connecting two Edge devices to each other with I2C... I think this is possible in theory but might be harder in practice. First of all I have simply not yet dived into the 'IOSlave' capabilities of the Apollo3. I know that it has the ability to act as a I2C/SPI slave device but I'm not sure what the hard parts will be yet. Also the IOSlave pins might not be exposed on the Edge board. It might be more practical for you to connect the two Edge boards with UART, both being able to send and receive.

As far as wrapping the Wire library in C I would suggest trying to just 'extract' the important parts from the Wire library, because it is already a C++ wrapper of the C HAL functions. (Its a funny thought to think of a C wrapper of a C++ wrapper of C.... like a C++ sandwich!)

Don't be afraid of the HAL, it is very useful and fairly well documented. I rely heavily on the features of Visual Studio Code to search for relevant functions, type definitions and the likes. I also usually spend an hour or two just reading the 'am_hal_xxx.h' files for whatever I'm interested in (like am_hal_iomaster.h) just to get familiar. Also there are usually some good examples in the AP3_EVB board examples folder (they can show you how to use the various HAL pieces).

Lastly, any code that you could write in the SDK can also be written in Arduino for rapid prototyping. You only need to keep in mind that some things might be going on behind the scenes.

Thanks and let us know how it goes!

RicoFio, 2019-07-26:
Hi @oclyke,

Haha that sandwich idea cracked me up lol. Awesome thanks a lot! I will dwell into it then and try to get it run. Also thanks for the hint with Visual Studios. I have to correct myself actually, I managed to get the Edge to communicate with a [SparkFun Pro RF](https://www.sparkfun.com/products/14916) over I2C where the RF board was configured as slave.
However, I'll give it a shot and keep you guys updated here!

RicoFio, 2019-07-31:
Hi @oclyke,

So the sandwich was disregarded and I tried to find the necessary bits and pieces in the Wire.cpp and the spi_testing_in_c.c (from [here](https://github.com/sparkfun/SparkFun_Edge_BSP/blob/master/examples/spi_testing_in_c/src/main.c)). What I ended up with is a pretty close solution but I am getting the retval **134217730** from the **am_hal_iom_blocking_transfer** function which I found originates from AM_HAL_IOM_ERR_I2C_NAK in **internal_iom_get_int_err()**.

Here the code I wrote so far [https://pastebin.com/i2SimsGg](ON PASTEBIN)

**My setup**: SparkFun Edge connected via QWIIC to a SparkFun Pro RF

The example code **Example5_Wire.ino** works perfectly and the Edge recognises the Pro RF as an I2C slave on address 0x04.

Do you have any idea what I am doing wrong? I followed the same setup process as yours in Wire.cpp and also the transmission is dispatched the same way.

Thanks a lot in advance!

oclyke, 2019-07-31:
Hmm, your code looks alright to me! Do you have access to a logic analyzer? My next step would be to inspect the communication lines and see if there is in fact a NAK condition being sent by the slave, and also whether the transmissions look any different between your code and the Wire implementation.

The master is supposed to send a NAK in some conditions (like after a multiple-byte read / write maybe) so could that be a factor? 

Maybe there is a difference with the ```.bContinue``` setting which may possibly confuse the slave?

oclyke, 2019-08-05:
@RicoFio I am cleaning up some old issues. How's this project coming? If you want to continue discussing it let's move to the SparkFun forums (where I will gladly hear more about your efforts!)

https://forum.sparkfun.com/viewforum.php?f=164

I'll close off this issue.

closed 2019-08-05 by oclyke