sparkfun/Arduino_Apollo3 #56: variant.h: No such file

gdsports, 2019-09-21:
### Subject of the issue
variant.h: No such file

### Your workbench
* Arduino IDE 1.8.10
* RedBoard Artemis

### Steps to reproduce
Compile Blink example with board type SparkFun RedBoard Artemis

### Expected behaviour
Successful compile.

### Actual behaviour
fatal error: variant.h: No such file or directory

Using version 1.0.9.

Three folders in the variant folder must be changed from BlackBoard to RedBoard to match the change to boards.txt in https://github.com/sparkfun/Arduino_Apollo3/commit/4ef0c020f9e8f6aedacae519c212a6de51121dbf

Without this change, compiling Blink.ino for SparkFun RedBoard Artemis* fails with "fatal error: variant.h: No such file or directory"

```
OLD:
variants/SparkFun_BlackBoard_Artemis/
variants/SparkFun_BlackBoard_Artemis_ATP/
variants/SparkFun_BlackBoard_Artemis_Nano/

NEW:
variants/SparkFun_RedBoard_Artemis/
variants/SparkFun_RedBoard_Artemis_ATP/
variants/SparkFun_RedBoard_Artemis_Nano/

```

nseidle, 2019-09-21:
Sorry about that. Oversight on our part. Fixed in [v1.0.10](https://github.com/sparkfun/Arduino_Apollo3/releases/tag/v1.0.10_ble-beta).

closed 2019-09-21 by nseidle

oclyke, 2019-09-23:
Thanks for fixing this @nseidle