sparkfun/Arduino_Apollo3 #58: RPi 4 / Apollo3 Board v 1.0.9 / arm-none-eabi-gcc not available

blu-raspberry, 2019-09-21:
### Apollo3 Board will not install from Board Manager on Pi 4. (Raspbian Buster)
The error is "Tool arm-none-eabi-gcc is not available for your operating system."

I have the 32 bit version of Arduino v 1.8.10 (64 bit is not applicable.)
I prefer using a Pi (3 or 4) for physical computing but in the case of my new Artemis RedBoard it seems to me that is not an option unless or until the board support is enhanced for this platform.

Please advise.  Thank you!

oclyke, 2019-09-23:
Thanks for your interest!

This is an error raised by Arduino trying to indicate that the dependencies cannot be fulfilled. The dependencies are [specified in the index JSON here](https://github.com/sparkfun/Arduino_Boards/blob/master/IDE_Board_Manager/package_sparkfun_index.json#L34). That tells Arduino "we need this tool"

And [later in the same file](https://github.com/sparkfun/Arduino_Boards/blob/master/IDE_Board_Manager/package_sparkfun_index.json#L1120) we say "here's where you can get that tool for various platforms!"

So really what is needed is to:
1. Find out what ```"host"``` value Arduino is looking for on your Pi 3/4 and then either
2. a. make a new entry with that host value and a link to one of the tools we already host or
2. b. get the proper arm-nbone-eabi (2018 q4 major) tools binary for the Pi and post it here so that we can host it - then use the new link to make the new entry supporting that tool for Pi. 

Unfortunately I just don't have time right now to find out the proper host key myself and I haven't seen any obvious documentation by Arduino. If you're willing to do it you could start by looking at the JSON package index for a core that does work on Pi.

oclyke, 2019-11-18:
Based on [this forum post](https://forum.sparkfun.com/viewtopic.php?f=164&t=50432) I am expanding this issue to general additional host support.

Goals:
- RPi (4, 3, Zero)
- Jetson Nano

Some host values to consider
- arm-linux-gnueabihf
- aarch64-linux-gnu

And also we may need to provide the bootloader binaries compiled for the respective architectures.

oclyke, 2020-01-06:
Related: #105

Wenn0101, 2021-07-20:
closed due to inactivity

closed 2021-07-20 by Wenn0101