sparkfun/Arduino_Apollo3 #65: pins_arduino.h not found

Mehuli-Ruh11, 2019-09-25:
### No pins_arduino.h in the variants
I am using red board Artemis nano with 2.8" adafruit TFT LCD shield. For simple graphic test, I am getting error 
`pins_arduino.h: No such file or directory`

###  workbench
* I am using Arduino IDE version 1.8.9
* The Arduino is connected as shown here: 
[https://learn.adafruit.com/adafruit-2-8-tft-touch-shield-v2/connecting](url)

### Expected behaviour
The LCD screen should show graphical outputs

### Actual behaviour

`Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "SparkFun RedBoard Artemis Nano, 115200, Ambiq Secure Bootloader (Advanced)"`

`C:\Users\Documents\Arduino\libraries\Adafruit_ILI9341\Adafruit_ILI9341.cpp:52:12: fatal error: pins_arduino.h: No such file or directory`
  ` #include "pins_arduino.h"`
         `   ^~~~~~~~~~~~~~~~`
`compilation terminated.`

oclyke, 2019-09-26:
Wow GitHub is getting smart... I just said "Fixes #65" in the PR and this issue got automatically closed... 

As pointed out by @nseidle  the ```pins_arduino.h``` file seems to be a predecessor to variant files. I took a glance at the Adafruit libraries and they seem to use the standard interfaces (digitalWrite, SPIClass etc...) so all we needed to do was include an empty ```pins_arduino.h```

I also had to add ```wiring_private.h``` and a weak definition for the ```yield()``` function. 

Your example should now compile. Please let us know if it works for you.

oclyke, 2019-09-26:
Oh.... to test this out you'll need to get v1.0.12 through the Arduino boards manager. I'll have it posted in under an hour

closed 2019-09-26 by oclyke

oclyke, 2019-09-26:
Ok, v1.0.12 is released w/ the fixes

Mehuli-Ruh11, 2019-09-27:
Now I am able to upload the sketch without any error but I am unable to see the desired graphical output.

oclyke, 2019-10-03:
I would recommend double checking your pin / SPI port assignments. Otherwise it would be best to check in with the creators of the TFT product that you are using because they will have better insight.

PR #66 (2019-09-26): Support Adafruit ILI9341 GFX
Fixes #65