sparkfun/Arduino_Apollo3 #69: Upload fails when connected to MAX30105 breakout board

Mehuli-Ruh11, 2019-10-02:
### Subject of the issue
I have connected a MAX30105 particle sensor breakout board to the QWIIC of the Redboard Artemis nano. I am uploading a sketch for simple serial reading. The upload fails every time.

### workbench
*Arduino 1.8.9
* Redboard Artemis Nano and MAX30105 particle sensor breakout board (https://www.sparkfun.com/products/14045)
* qwiic connector: black- GND, red- 5V, blue- SDA, yellow- SCL 
*Libraries used: https://github.com/sparkfun/SparkFun_MAX3010x_Sensor_Library

### Steps to reproduce
    `#include <Wire.h>
     #include "MAX30105.h"

     MAX30105 particleSensor;

    #define debug Serial //Uncomment this line if you're using an Uno or ESPww  d3
    //#define debug SerialUSB //Uncomment this line if you're using a SAMD21

    void setup()
    {
    debug.begin(115200);
    if (particleSensor.begin() == false)
    {
    debug.println("MAX30105 was not found. Please check wiring/power. ");
    while (1);
    }
    byte ledBrightness = 0x1F; //Options: 0=Off to 255=50mA
    byte sampleAverage = 2; //Options: 1, 2, 4, 8, 16, 32
    byte ledMode = 3; //Options: 1 = Red only, 2 = Red + IR, 3 = Red + IR + Green
    int sampleRate = 400; //Options: 50, 100, 200, 400, 800, 1000, 1600, 3200
    int pulseWidth = 69; //Options: 69, 118, 215, 411
    int adcRange = 4096; //Options: 2048, 4096, 8192, 16384
  
    particleSensor.setup(ledBrightness, sampleAverage, ledMode, sampleRate); //Configure 
    sensor. 
    Use 6.4mA for LED drive
    }
    void loop()
    {
    debug.print(millis());
    debug.print(",");
    debug.print(particleSensor.getRed());
    debug.print(",");
    debug.print(particleSensor.getIR());
    debug.print(",");
    debug.print(particleSensor.getGreen());
    debug.println();
    }`

### Expected behaviour
The expected output should be serial reading on the serial monitor

### Actual behaviour
`
    Connecting over serial port COM16...
    Fail
    Fail
    Fail
    Tries = 3
    Upload failed
`

oclyke, 2019-10-03:
@Mehuli-Ruh11 does the upload fail or succeed when the MAX30105 is not connected?

Mehuli-Ruh11, 2019-10-03:
@oclyke No, it does not. It uploads but in the serial monitor I get output that,
`MAX30105 was not found. Please check wiring/power.`

nseidle, 2019-10-03:
Your code compiles and runs fine for me:

[image]

I tested both Artemis RedBoard and Artemis ATP.

I am confused by your report: You say in OP that upload fails. Ok, that's one issue. Then you say " It uploads but in the serial monitor I get output that, 'MAX30105 was not found. Please check wiring/power.'"

So it seems you can upload but then com with MAX30105 fails. I suspect your hardware connections may be a problem. Plug the MAX30105 back in. Be sure the solder connections are clean and that SCL is indeed yellow.

If you still have issues, please let us know.

nseidle, 2019-10-03:
BTW - I'm on windows 10. What is your OS out of curiosity?

Mehuli-Ruh11, 2019-10-03:
I am windows 10 as well. If I do not connect with the qwiic connector but manually solder with the SDA3, SCL3, GND, and 3v3 should there be any change to the Arduino code? I am changing the initial part as 

[image]

It gives the same upload failed error!

Mehuli-Ruh11, 2019-10-03:
@nseidle Also, the upload fails for me when the MAX30105 is connected but the sketch uploads to the redboard with the 'MAX30105 not found' error when the MAX30105 is not connected.

nseidle, 2019-10-03:
Ah, you're using port 0. Yep, works fine. Make sure yellow Qwiic wire goes to SCK, blue wire to MISO.

[image]

This is not an Artemis issue. Is the power LED on your RedBoard turning off when you attach your MAX30105 board? You've got a heck of a hardware issue. Either something is wrong with your MAX30105 board or there's a problem with the connections to Artemis.

Mehuli-Ruh11, 2019-10-04:
My bad, in my code it is 
`TwoWire myWire(3)`
when connected to SDA3 and SCL3.
The led on the redboard does not turn off when connected to the  MAX30105. The MAX30105 works fine with arduino nano ATMEGA328.

closed 2019-10-07 by Mehuli-Ruh11