sparkfun/Arduino_Apollo3 #74: Windows recognizes artemis_svl.exe as a virus

JordanMajd, 2019-10-14:
### Issue

First time I tried to compile &  upload example `Example8_BLE_LED` after installing core via the Arduino IDE:

```bash
CreateProcess error=225, Operation did not complete successfully because the file contains a virus or potentially unwanted software at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
```

Then Windows removes the file from the system. If you try to run again it fails to locate the file:

```bash
java.io.IOException: Cannot run program "C:\Users\lordbunson\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.15/tools/artemis/windows/artemis_svl.exe": CreateProcess error=2, The system cannot find the file specified
```

### Details

- Platform: Windows 10 Pro Version 1809 OS build 17763.805
- Device: Artemis Nano
- How is device wired: USB C to  USB C
- How is everything being powered? USB C
- Are there any additional details that may help us help you?
    - If you try to click on the file after a fresh install a dialog pops up with the original error, then the file is removed from the system.

### Repro

From git bash or WSL:

```bash
# backup Arduino installed board
mv ~/AppData/Local/Arduino15/packages/SparkFun/hardware/apollo3/1.0.15 1.0.15.bak
# clone from GH
git clone https://github.com/sparkfun/Arduino_Apollo3.git ~/AppData/Local/Arduino15/packages/SparkFun/hardware/apollo3/1.0.15
~/AppData/Local/Arduino15/packages/SparkFun/hardware/apollo3/1.0.15/tools/artemis/windows/artemis_svl.exe
```

Press `Win+R` to open up run, enter `%userprofile%/AppData/Local/Arduino15/packages/SparkFun/hardware/apollo3/1.0.15/tools/artemis/windows/` to open the directory in explorer.

Now you can double click the file to see the error dialog watch it get deleted in real time.

It's easy to rinse and repeat by simply running a `git checkout tools` to bring the exe back.

This happens regardless of whether you install via the official `package_sparkfun_index.json` via boards manager or by cloning from GH. It also does not seem to matter what branch I am on.

oclyke, 2019-10-14:
@JordanMajd Thanks a bunch for the report. We are looking into this - it seems to have cropped up recently without any apparent cause. As a stop-gap fix I am making a new release that contains an archived (v1.0.12) version of the uploader tool.

JordanMajd, 2019-10-14:
Thanks! That fixes it.

I removed my apollo3 boards and installed v1.0.12 and was able to upload just fine.

Additionally, I was able to use the v1.0.12 `artemis_svl.exe` on other releases / branches successfully:

```bash
cd ~/AppData/Local/Arduino15/packages/SparkFun/hardware/apollo3/
git clone -b core-ble https://github.com/sparkfun/Arduino_Apollo3.git 1.0.15
cp  1.0.12/tools/artemis/windows/artemis_svl.exe 1.0.15/tools/artemis/windows/artemis_svl.exe
```

oclyke, 2019-10-14:
Glad to hear it! I've pushed the release to Arduino so it should be available through the boards manager soon (v1.0.16). I will keep this issue open until we can fix the underlying problem.

Notes for later:
Some part of the combination of Windows, pyinstaller, and the source code (and perhaps packages used in the source?) is triggering a virus detection. v1.0.12 seems to work fine.

oclyke, 2019-10-30:
Probably fixed by #81. Closing. Please feel free to re-open if there are any more problems.