sparkfun/Arduino_Apollo3 #83: Using Visual Studio Code for Apollo3

justiceamoh, 2019-10-30:
### Subject of the issue
I'm trying to setup a development environment in VS Code for the Apollo 3 (Sparkfun Redboard Artemis) and I was hoping someone could point me in a good starting direction. @oclyke, you implied in issue #36 that you use VS Code. What is your setup? Do you have the Ambiq SDK setup to compile in VSCode or do you use the arduino-vscode extension? The extension is limiting for me -- I cannot jump into HAL function definitions or debug in general. If you have a sample VS Code project, that will be very helpful.

**Some background**: I am successfully compiling AmbiqSDK examples and my own projects using Makefiles and the command line tools. I have also (with help from Sparkfun Forums) been able to setup a Segger Embedded Studio (SES) for the Apollo3/Artemis and can debug code using J-LINK.

### Your workbench
macOS 10.15, Catalina
Sparkfun Redboard Artemis
USB-C, J-LINK Debugger
Powered via USB

oclyke, 2019-10-30:
@justiceamoh I use a pretty vanilla install of VSCode - I only have things like the C/C++ and Python linters installed. I often use the integrated terminal but I don't use, for example, the arduino-vscode or debugging extensions. Once upon a time I tried to get debugging working but it ended up being easier to use SES. I might come back to it. I also could consider using the arduino extension depending on what it does. But yeah I hope that answers your question - I haven't really done anything super special with VSCode.

closed 2019-10-30 by oclyke

oclyke, 2019-10-30:
Hmm... Ive never personally configured anything like ```c_cpp_properties.json```.  I'll be honest my SES setup is incredibly basic and only useful in emergencies when I need a debugger - ultimately I would like to work out a better solution but can't justify it right now. 

My basic strategy is to make sure my SES project is located at the root of whatever project I'm working on (e.g. tensorflow or something) so that all the relative paths can be worked out by SES. Other than that I am leaving it up to the black magic of SEGGER.

justiceamoh, 2019-10-30:
Ah I see. Thanks for letting me know. 

Btw, If you want a more revamped SES setup, check out [this post on the Sparkfun forum](https://forum.sparkfun.com/viewtopic.php?f=171&t=50960). I've been using the the setup guide the OP shared and it's quite comprehensive.

oclyke, 2019-10-31:
@justiceamoh Thanks for the link. I also just got a new debug setup working from this article by SEGGER:
https://wiki.segger.com/J-Link:Visual_Studio_Code

I'm not sure how long it has been available but it seems t be working for me!

justiceamoh, 2019-11-05:
Actually, I just debugged with the `.axf` file and it worked ok. Thanks!

oclyke, 2019-11-06:
Perfect! I'm no expert but from my experience ```.axf``` can be used wherever a ```.elf``` was called for.

justiceamoh, 2019-11-06:
Awesome. That's good to know going forward. Thanks for getting back.