sparkfun/Arduino_Apollo3 #97: package.json - working towards Issue #14

nigelb, 2019-12-10:
### Working towards #14 - Support for PlatformIO
I have been playing around using this repository as a framework to build with platformio, see [platform-apollo3blue](https://github.com/nigelb/platform-apollo3blue). To use this repo as a framework package for platformio it needs a simple change, the addition of a `package.json` file containing a small amount of metadata, and looks like this:

```json
{
    "name": "framework-arduinoapollo3",
    "description": "Arduino Wiring-based Framework (Apollo3 Core)",
    "version": "1.0.21",
    "url": "https://github.com/sparkfun/Arduino_Apollo3"
}
```

oclyke, 2019-12-10:
Neat! If we include this as-is what level of support for PlatformIO does that give us? We'd be happy to get a PR to recognize your contribution too!

nigelb, 2019-12-13:
Hi @oclyke currently it compiles and uploads code fine. Since I have can now upload I have been testing out the provided examples and have been running into issues in which the program seems to hang midway through. I am not sure how to solve this issue without attaching a debugger and stepping through the code. I have been documenting the issues [Here](https://github.com/nigelb/platform-apollo3blue/blob/master/Tests.md), any suggestions would be welcome. 

I can make a pull request with the above file, if that is what you mean.

oclyke, 2019-12-13:
Yeah I'd appreciate the PR (I know that I could just copy/paste that JSON but it's your work and it represents effort to learn about PlatformIO [which I haven't done] and we appreciate it.)

Could you also give me an idea of the basics for how to use PlatformIO? Then I could see if I can tell why the programs are hanging part way through. Thanks!

nigelb, 2019-12-14:
I have made the pull request. 

Here is a quick into to PlatformIO. Hopefully I have covered enough to get you up and running.
## Basic PlatformIO Install
Although it supports a bunch of other IDEs, I will just deal with their preferred option:

1. [Instillation instructions](https://platformio.org/install/ide?install=vscode)
2. A quick introduction [Quickstart](https://docs.platformio.org/en/latest/ide/vscode.html#quick-start)

The IDE installs PlatformIO to `C:\Users\<USERNAME>\.platformio\penv\Scripts\platformio.exe`  which can be used in the Terminal window of the IDE or from a CLI. It is also simple to install PlatformIO into a python virtualenv as well: `pip install -U platformio` ([Docs](https://docs.platformio.org/en/latest/installation.html#python-package-manager)).

Once you have the IDE and PlatformIO installed we need to install the Apollo3 Platform.

## Install platform-apollo3blue
Since the `platform-apollo3blue` platform in not in the PlatformIO registry, we need to do a [manual install](https://github.com/nigelb/platform-apollo3blue/blob/master/README.md). The instructions are for Linux, let me know if you have any issues transferring them over to Windows. Mostly it is just replacing the `platformio` commands with `C:\Users\<USERNAME>\.platformio\penv\Scripts\platformio.exe`

## Testing the Examples
Once you have all of that installed create a new project and select which ever Artemis board you are testing with (I have been using the SparkFun_Artemis_Nano)

For the [tests](https://github.com/nigelb/platform-apollo3blue/blob/master/Tests.md) I did, after creating a new project I opened the example in the Arduino IDE and copied the code, then in the PlatformIO IDE I created a new `cpp` file in the `src` directory and pasted in the example code. The only changes I made to this code was to add the missing function prototypes and maybe add a `#include "Arduino.h"` if required.

Once that is done you can press the upload button (which was illustrated in the [Quick Start Guide](https://docs.platformio.org/en/latest/ide/vscode.html#quick-start)) and then open the Serial Monitor to check the results.

Also, to get verbose output for the build you may need to copy the command and add the `-v` command line flag.

## Modifying the build flags
The build flags passed to the compiler can be modified by editing two files:
- [main.py](https://github.com/nigelb/platform-apollo3blue/blob/master/builder/main.py) `~/.platformio/platforms/apollo3/builder/main.py`
- [arduino.py](https://github.com/nigelb/platform-apollo3blue/blob/master/builder/frameworks/arduino.py) `~/.platformio/platforms/apollo3/builder/frameworks/arduino.py`

The `arduino.py` is the main one however.

justiceamoh, 2020-01-17:
I can help out with this. I have 'some' experience with PlatformIO so I will take a stab at testing the PR later today. Thanks for putting this together @nigelb.

oclyke, 2020-01-17:
@nigelb Great instructions! I haven't had time to try it out yet - sorry. 
@justiceamoh that would be highly appreciated. 

Thanks to both of you!

justiceamoh, 2020-01-18:
I can confirm that @nigelb's framework for the artemis boards works well. I tested with the Redboard Artemis Nano on a mac, macOS 10.15. I caught a bug that was preventing upload to the target specifically on a macOS machines -- basically, the path to `artemis_svl` was incorrectly set by the builder script. It was a quick fix though and I've submitted [a PR](https://github.com/nigelb/platform-apollo3blue/pull/2) for it. 

Other things worth mentioning, once I installed both the Sparkfun Arduino_Apollo3 and the platform-apollo3blue, I had to run `pio platform update` before the board was visible in PlatformIO's boards. Also, once the board was installed, I completed setting up the project using the desktop version of PlatformIO (as a plugin in VS Code). And that worked very well. Upon creating a new project, the `main.cpp` is provided with the expected `setup()` and `loop()`, as well as an included `Arduino.h`. 

One feature that might be good to add is to automate the inclusion of `upload_port=` and `upload_speed=` in the platform.ini file. I can look into those. 

Thanks again for the great work @nigelb! I will keep testing and will suggest improvements if I come up with any.

justiceamoh, 2020-01-20:
I just submitted another [PR](https://github.com/nigelb/platform-apollo3blue/pull/3) for automatic detection of serial port name so that users don't have to manually define the `upload_port` or the `upload_speed` in the `platform.ini` file. Ofcourse, users can still chose to if they so desire.

I also see you've had issues with the PDM and EEPROM examples. I have some experience with PDM so I will take a stab at that next.

nigelb, 2020-01-28:
Thank @justiceamoh I have accepted those pull requests.

jerabaul29, 2020-12-05:
Just curious, what is the status on this? :) .