sparkfun/Arduino_Apollo3 #106: Support Binary Assignment to Variables

Marsfan, 2020-01-12:
### Subject of the issue
Arduino supports assigning values to variables in binary form by prefixing the number with a capital `B`.  (Ex: `n = B101`)
When I try to compile a program for my Artemis board, this fails, saying that the value was not declared in this scope. 

### Your workbench
* Arduno 1.8.10 on Windows 10 1909 x64
* Using Artemis ATP
* Connected via USB-C cable to laptop USB-C port
* Powered over USB-C

### Steps to reproduce
1. Create a new Arduino project [with this code](https://gist.github.com/Marsfan/aea6a9f26a6c26ffbad69e1146444296). 
2. Configure the Arduino IDE to use the Redboard Artemis ATP
3. Verify the program.

### Expected behaviour
Program verifies successfully. 

### Actual behaviour
Verification fails with the error that `'B101 was not declared in this scope'`

Marsfan, 2020-01-13:
Ok. I think I sort of figured it out. 
For the Artemis, binary values are written with a 0b prefix, such as `0b101`. 
It might still be nice to support the traditional method though, for compatibility with other boards.

afoxinsocks, 2020-01-13:
> Ok. I think I sort of figured it out.
> For the Artemis, binary values are written with a 0b prefix, such as `0b101`.
> It might still be nice to support the traditional method though, for compatibility with other boards.

0b is the correct prefix for binary numbers in C++ (which Arduino is based on) and C (later versions like C11 support it, I believe), like 0x is the correct prefix for hexadecimal numbers and 0 alone is the octal prefix. I do not believe B has ever been allowed as a prefix for defining a numeric literal, because a requirement for variable naming in C/C++ is that they start with a letter and can never start with a number.

Marsfan, 2020-01-13:
@afoxinsocks Understood. But as per the link I attached, the Arduino IDE _does_ support the form `B101`. It is documented as such by Arduino, and I have found other code that uses it.

afoxinsocks, 2020-01-13:
@Marsfan The gist you linked is not from official Arduino code and probably doesn't compile at all on any platform because of restriction for variable naming/numeric literals I mentioned -- if a token starts with a letter, it is evaluated by the compiler to be a variable, not a literal. 
If I'm wrong about this... well.. at least they fixed it :) Consistency is good and it's good for Arduino to be C++ at its core. Have a good one!

Marsfan, 2020-01-13:
My apologies. I forgot to add a second link into my original post. 
See here: 
[https://www.arduino.cc/reference/en/language/variables/constants/integerconstants/](https://www.arduino.cc/reference/en/language/variables/constants/integerconstants/)

afoxinsocks, 2020-01-13:
I'm inclined to think that Arduino page is simply wrong because of the earlier comments about literal/variable naming requirements in C/C++. , and the code samples you've seen using B as a binary prefix probably don't compile either and left a lot of people wondering what is going on. Then again, I've seen a lot of C/C++ programmers that never use the 0b prefix, favoring 0x instead, so maybe not that many people have been bitten by this. Going back to the earliest days of C, variables have always started with a letter, which is also probably why the syntax highlighter on that linked page above isn't coloring the "B101" the same way it colors the other literals (orange).

Marsfan, 2020-01-13:
@afoxinsocks 
I just tested. The strange way of writing binary (`B101`) works for compiling for the Arduino UNO. It looks like it is some sort of non-standard Arduino thing.

nseidle, 2020-01-13:
@afoxinsocks - You're totally right, Arduino is weird and has some custom B### nomenclature. I'd normally forget it but we want to support as much code 'out of the box' as possible. I don't want customers to have to think, it should just compile. And since we have control over the core, we'll try to support as much reverse compatibility as possible.

@Marsfan - Thanks for reporting! 

If anyone has time for a PR or recommended solution please do so. This issue needs fixing but is not a high prio for us right now.

oclyke, 2020-01-13:
Just chiming in that I'm along for the ride - we will find a solution for this. Can't get to it right away but ASAP.

oclyke, 2020-01-17:
So I have been thinking about this one... It is funny that the [Arduino documentation](https://www.arduino.cc/reference/en/language/variables/constants/integerconstants/) mentions a restricted domain for these values [0, 255]

Maybe they've just fudged it with a  bunch of macros?

oclyke, 2020-02-06:
This should be supported in the next release.

closed 2020-02-06 by oclyke