sparkfun/Arduino_Apollo3 #109: Arduino core 1.0.25 broken Example8_BLE compile can't find ap3_g_prfbuf

kerryeven, 2020-01-16:
### Subject of the issue
Describe your issue here.
compile error: 
C:\Users\kerry\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.25\cores\arduino\ard_sup\uart/ap3_uart.cpp:426: undefined reference to `ap3_g_prfbuf'
### Your workbench
* What platform are you using? Windows 10
* What version of the device are you using? Is there a firmware version?Artemis Redboard Nano
* How is the device wired to your platform? not connected..just Verify/Compile
* How is everything being powered?
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue to a gist. Using standard example Example8_BLE .. same error with Example5_analogWrite

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

oclyke, 2020-01-16:
Thanks for the report. This is my fault - working on a fix as we speak.

kerryeven, 2020-01-16:
thanks...26 works...that was fast...

oclyke, 2020-01-16:
26 compiles but does not work... there is something more sinister going on here haha

kerryeven, 2020-01-16:
Yep...26 no worky...my robot has 3 motors, 1 servo .. see it in BLE but no motor control anymore.
re-install 24...back to working.  fyi...

oclyke, 2020-01-17:
Closing in favor of #110

closed 2020-01-17 by oclyke