sparkfun/Arduino_Apollo3 #110: Serial no longer functional with Arduino Core for Apollo 3 v1.0.26

adamgarbo, 2020-01-17:
### Subject of the issue
- Arduino Core for Apollo 3 v1.0.26 appears to have broken Serial output

### Your workbench
- What platform are you using?
    - Software: macOS Catalina, Arduino IDE 1.8.10
    - Hardware: SparkFun Artemis Nano, Edge 2
- What version of the device are you using? Is there a firmware version?
    - Arduino Core for Apollo 3 v1.0.26
- How is the device wired to your platform?
    - USB C/FTDI
- How is everything being powered?
    - USB
- Are there any additional details that may help us help you?
    - I'm unsure of the nature of this issue, but after updating to v1.0.26 all things Serial related ceased working. When I rollback to v1.0.24 there's no problems at all with Serial. Code that does not involve Serial appears to be working without issue (blink).

### Steps to reproduce
- Uploaded the Example4_analogRead and Example1_getTime

### Expected behaviour
- Visible output on Serial Monitor

### Actual behaviour
- No output on Serial Monitor
- In the analog read example, an LED should blink but this is not observed. Possibly indicative that code is not executing beyond Serial.begin()

oclyke, 2020-01-17:
Thanks - working on a fix. Not far out now

oclyke, 2020-01-17:
This should be resolved w/ the 1.0.27 release. Sorry about the roller coaster ride everyone ;)

You can test the latest Serial improvements / functionality with this example:

``` c
#define BAUD_RATE 9600

void setup() {
  Serial.begin(BAUD_RATE);
  delay(10);
  Serial.println("Serial buffer testing");

  String myString = "Lorem ipsum dolor sit amet, consectetur adipiscing elit. Pellentesque eget ligula laoreet, faucibus felis sit amet, cursus mi. Praesent congue tristique congue. Pellentesque feugiat diam volutpat eros consectetur dapibus. Nullam lobortis enim at rutrum maximus. Sed lobortis purus in sagittis.";

  long myTimer = millis();
  Serial.println(myString);
  long delta = millis() - myTimer;
  
  Serial.println("Length: " + (String)myString.length());
  Serial.println("Time to send myString: " + (String)delta + " ms");
  Serial.println("Time to send with blocking method (calculated): " + (String)(myString.length() * (8.0/BAUD_RATE) * 1000) + " ms");
  Serial.print("Available space in TX buffer: ");
  Serial.println(Serial.availableForWrite());
  Serial.println("\nEcho:");
}

void loop() {
  while(Serial.available())
  {
    Serial.write(Serial.read());
  }
}
```

closed 2020-01-17 by oclyke