sparkfun/Arduino_Apollo3 #114: Conflict with arm_math.h

nakanaela, 2020-01-24:
I have tried to compile Example1_MicrophoneOutput for the Artemis Thing Plus and the compilation failed with the following errors

```
In file included from /Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/ard_sup/Arduino.h:86,
                 from sketch/Example1_MicrophoneOutput.ino.cpp:1:
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/variants/artemis_thing_plus/config/variant.h:51:12: error: expected unqualified-id before numeric constant
 #define A0 19
            ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h:1720:11: note: in expansion of macro 'A0'
     q15_t A0;           /**< The derived gain, A0 = Kp + Ki + Kd . */
           ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/variants/artemis_thing_plus/config/variant.h:52:12: error: expected unqualified-id before numeric constant
 #define A1 20
            ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h:1725:11: note: in expansion of macro 'A1'
     q31_t A1;           /**< The derived gain A1 = -Kp - 2Kd | Kd.*/
           ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/variants/artemis_thing_plus/config/variant.h:51:12: error: expected unqualified-id before numeric constant
 #define A0 19
            ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h:1738:11: note: in expansion of macro 'A0'
     q31_t A0;            /**< The derived gain, A0 = Kp + Ki + Kd . */
           ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/variants/artemis_thing_plus/config/variant.h:52:12: error: expected unqualified-id before numeric constant
 #define A1 20
            ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h:1739:11: note: in expansion of macro 'A1'
     q31_t A1;            /**< The derived gain, A1 = -Kp - 2Kd. */
           ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/variants/artemis_thing_plus/config/variant.h:53:12: error: expected unqualified-id before numeric constant
 #define A2 21
            ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h:1740:11: note: in expansion of macro 'A2'
     q31_t A2;            /**< The derived gain, A2 = Kd . */
           ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/variants/artemis_thing_plus/config/variant.h:51:12: error: expected unqualified-id before numeric constant
 #define A0 19
            ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h:1752:15: note: in expansion of macro 'A0'
     float32_t A0;          /**< The derived gain, A0 = Kp + Ki + Kd . */
               ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/variants/artemis_thing_plus/config/variant.h:52:12: error: expected unqualified-id before numeric constant
 #define A1 20
            ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h:1753:15: note: in expansion of macro 'A1'
     float32_t A1;          /**< The derived gain, A1 = -Kp - 2Kd. */
               ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/variants/artemis_thing_plus/config/variant.h:53:12: error: expected unqualified-id before numeric constant
 #define A2 21
            ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h:1754:15: note: in expansion of macro 'A2'
     float32_t A2;          /**< The derived gain, A2 = Kd . */
               ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h: In function 'float32_t arm_pid_f32(arm_pid_instance_f32*, float32_t)':
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/variants/artemis_thing_plus/config/variant.h:51:12: error: expected unqualified-id before numeric constant
 #define A0 19
            ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h:4783:15: note: in expansion of macro 'A0'
     out = (S->A0 * in) +
               ^~
In file included from /Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/libraries/PDM/examples/Example1_MicrophoneOutput/Example1_MicrophoneOutput.ino:28:
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h:4783:15: error: expected ')' before numeric constant
     out = (S->A0 * in) +
           ~   ^
               )
In file included from /Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/ard_sup/Arduino.h:86,
                 from sketch/Example1_MicrophoneOutput.ino.cpp:1:
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h: In function 'q31_t arm_pid_q31(arm_pid_instance_q31*, q31_t)':
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/variants/artemis_thing_plus/config/variant.h:51:12: error: expected unqualified-id before numeric constant
 #define A0 19
            ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h:4818:22: note: in expansion of macro 'A0'
     acc = (q63_t) S->A0 * in;
                      ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/variants/artemis_thing_plus/config/variant.h:52:12: error: expected unqualified-id before numeric constant
 #define A1 20
            ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h:4821:23: note: in expansion of macro 'A1'
     acc += (q63_t) S->A1 * S->state[0];
                       ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/variants/artemis_thing_plus/config/variant.h:53:12: error: expected unqualified-id before numeric constant
 #define A2 21
            ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h:4824:23: note: in expansion of macro 'A2'
     acc += (q63_t) S->A2 * S->state[1];
                       ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h: In function 'q15_t arm_pid_q15(arm_pid_instance_q15*, q15_t)':
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/variants/artemis_thing_plus/config/variant.h:51:12: error: expected unqualified-id before numeric constant
 #define A0 19
            ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h:4870:40: note: in expansion of macro 'A0'
     acc = (q31_t) __SMUAD((uint32_t)S->A0, (uint32_t)in);
                                        ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/variants/artemis_thing_plus/config/variant.h:52:12: error: expected unqualified-id before numeric constant
 #define A1 20
            ^~
/Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include/arm_math.h:4874:40: note: in expansion of macro 'A1'
     acc = (q63_t)__SMLALD((uint32_t)S->A1, (uint32_t)*vstate, (uint64_t)acc);
                                        ^~
Multiple libraries were found for "PDM.h"
 Used: /Users/nbps/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.0.27/libraries/PDM
exit status 1
Error compiling for board SparkFun Artemis Thing Plus.
```

I've made no changes to the example, I installed SparkFun Apollo3 Boards (1.0.27) via the board manager, opened the example and tried to compile. It appears the conflict comes from the way the _arm_pid_instance_xxx_ are defined starting at line 1718. The derived gain variables are declared as A0, A1, and A2 which does not play well with the analog pin definitions in variant.h.

I am not sure if this can be easily remedied by changing the gain variables to, say, B0-B2, but as it stands, the example will not compile and I haven't tried it yet, but I believe anything using arm_math.h will have the same issues.

nseidle, 2020-01-24:
This was fixed previously. Looks like the recent SDK update overwrite the changes. Gimme one second. It's just renaming the vars in the header...

nseidle, 2020-01-24:
PR is [here](https://github.com/sparkfun/Arduino_Apollo3/pull/115). We'll get it released shortly. 

Thanks for reporting!

closed 2020-01-24 by nseidle

oclyke, 2020-01-24:
Core version 1.0.28 is released and available through Arduino. Fixes this issue. Thanks again!

nakanaela, 2020-01-25:
Thanks for the amazing fast response. Everything compiles as expected now. And it may have broken things for a bit, but I am glad to see that the included bits of the CMSIS library are being updated. Now to see if I can't figure out how to get FastLED working with the Apollo3 core.

nseidle, 2020-01-25:
We would *LOVE* help getting FastLED working on Artemis. We've looked into it but with the sad news around the FastLED repo owner we weren't sure we could get a PR accepted. Looks like [it's moving again](https://github.com/FastLED/FastLED/commits/master).