sparkfun/Arduino_Apollo3 #118: RTC month referencing is incorrect

stephenf7072, 2020-01-27:
### Subject of the issue
I'm using the Artemis RTC and it would sometimes get the number of days in a month wrong.

I've found what seems to be a bug in  Arduino_Apollo3/libraries/RTC/src/RTC.cpp, here are the problem lines and the fix:
Change line 58
From: `hal_time.ui32Month = month - 1; //HAL is expecting 0 to 11 months`
To: `hal_time.ui32Month = month;`

Change line 95
From: `month = hal_time.ui32Month + 1; //Convert from 0-11 to 1-12`
To: `month = hal_time.ui32Month;`

I'm not sure where, but it seems the 0-11 to 1-12 conversion is already done elsewhere.

### Steps to reproduce
Here's a test function to see the days go by
`#include "RTC.h"
#include <Time.h>
#include <TimeLib.h>

void timeWarp()
{  
  myRTC.setTime(23,59,59,99, 1,1,19); //Manually set RTC
  printArtemisTime();
  Serial.println();
  
  while(1)
  {
    time_t tempTime = readArtemisTimeNow();    
    Serial.print("Artemis:"); printArtemisTime();
//    Serial.print("\tTranslated:");
//    printTime(tempTime);
    Serial.println();
    
    myRTC.getTime();
    myRTC.setTime(23,59,59,98, myRTC.dayOfMonth,myRTC.month,myRTC.year); //Manually set RTC
    delay(200); 
  }
}

void printArtemisTime()
{
  myRTC.getTime();
//  Serial.print("Time(elements):");
  char buf[50];
  char weekdayBuf[4];
//  time_t tempTime = makeTime(theTime);
  int i = myRTC.weekday+1;
  switch(i)
  {
    case (1):
    {
      strcpy(weekdayBuf,"Sun");
      break;
    }
    case (2):
    {
      strcpy(weekdayBuf,"Mon");
      break;
    }
    case (3):
    {
      strcpy(weekdayBuf,"Tue");
      break;
    }
    case (4):
    {
      strcpy(weekdayBuf,"Wed");
      break;
    }
    case (5):
    {
      strcpy(weekdayBuf,"Thu");
      break;
    }
    case (6):
    {
      strcpy(weekdayBuf,"Fri");
      break;
    }
    case (7):
    {
      strcpy(weekdayBuf,"Sat");
      break;
    }

    default:
    {
      strcpy(weekdayBuf,"???");
      break;
    }
  }

//  myRTC.seconds,myRTC.minute,myRTC.hour,0,myRTC.dayOfMonth,myRTC.month,CalendarYrToTm(myRTC.year+2000)
  sprintf(buf,"%02d-%02d-%02d (%s) %02d:%02d:%02d.%02d", myRTC.year, myRTC.month, myRTC.dayOfMonth, weekdayBuf, myRTC.hour, myRTC.minute, myRTC.seconds, myRTC.hundredths);
  Serial.print(buf);
}`

oclyke, 2020-01-27:
@nseidle is more familiar with the RTC - but given a glance at the issue I would guess that this could be related to the update of the underlying HAL functions. We recently migrated to release 2.2.0 from 2.0.0.

nseidle, 2020-01-28:
Thanks for reporting @stephenf7072! I built Example3 into the RTC folder based on your code. [This PR](https://github.com/sparkfun/Arduino_Apollo3/pull/120) should fix the issue.

closed 2020-01-28 by nseidle