sparkfun/Arduino_Apollo3 #123: Arduino_TensorFlowLite hello_world example not compiling

vilaksh01, 2020-02-05:
### Arduino_TensorFlowLite hello_world example not compiling for Sparkfun Redboard Artemis ATP
I tried to upload the hello-world example after adding the below lines in start of program and changing different versions of tensorflow lite libraries available  
```
#if defined(min)
#undef min
#endif

#if defined(max)
#undef max
#endif

```
I am attaching the error which was generated by the compiler.
[CompilerWarnings.txt](https://github.com/sparkfun/Arduino_Apollo3/files/4160793/CompilerWarnings.txt)

oclyke, 2020-02-06:
What version of the Apollo3 core are you using? I just successfully compiled the Arduino_TensorFlowLite 'hello_world' example for RedBoard Artemis ATP using Arduino Apollo3 core version 1.0.28.

vilaksh01, 2020-02-06:
Sorry it worked, I just included a file named arduino command responder with ambsp inclusion for redboard and it worked

closed 2020-02-06 by vilaksh01

oclyke, 2020-02-06:
@vilaksh01 Thanks for reporting this. I am a little confused - did you have to make modifications to get this working? Or was your platform set up differently than a stock 1.0.28 installation to begin with?

If there is something that needs to be changed to make this work out of the box we would definitely like to fix it.