sparkfun/Arduino_Apollo3 #128: SDIO 4 bit mode for Sd card and low power

JPB1JPB1, 2020-02-17:
Dear Sparkfun team

Can the Apollo 3 supports SDIO 4 bit mode for SD/uSD card ? 
Library code and pinout ?
At the same time, can it be optimized for Speed and Power consumption using speed/size of buffer, "smallest fat" support footprint ...

Please, example are requested..
Best Regards
Jp

nseidle, 2020-02-17:
We believe the Artemis/Apollo3 can support SDIO mode. The Apollo3 has 1/2/4/8 SPI so the hardware is there. Please see the Apollo3 datasheet for pad mapping. I made this [single page reference](https://github.com/sparkfun/SparkFun_Artemis/blob/master/Documents/Apollo3%20Pad%20Mapping.pdf) if that helps. You need pads 22, 26, 4, 23 (MSPI0-3), with CLK on pad 24. You'll also need a CE but there's a bunch of pads available:

[image]

In the community, Bill Greiman has been doing a large amount of work to support SDIO on Teensy with [sdfat-beta](https://github.com/greiman/SdFat-beta). I haven't proven out anything but I suspect the crossover should not be too daunting.

So for now, there is no support for SDIO but the hardware support is there and we're very interested in implementing it in a future project/product.

JPB1JPB1, 2020-02-18:
Thanks for your reply and good information..

But since Apollo 1-2 and now 3 , there is really no code  for SDIO 4 bit mode support or beta- example ?
One issue is that there is NO sdio-4-bit mode in Arduino library... 

Please, any timeframe when it can be test/use/certifity ? 
Have a great day,
Best Regards
Jp

idea--list, 2020-02-18:
Thing is the SPI solution is free of charge, much easier to implement the code but slow.
If i am right for SDIO4 one needs to pay some licence costs and is also harder to implement, as a consequence of these it is not easy to find SDIO4 implementations.
[This however might give some inspiration](https://github.com/JojoS62/COMPONENT_SDIO) (even if it is written for mbed)

oclyke, 2020-05-05:
Closing because Apollo3 does not have a dedicated 4-bit SDIO peripheral. Support for this mode is outside the scope of the core and should be a library instead

closed 2020-05-05 by oclyke