sparkfun/Arduino_Apollo3 #131: ADC readings when multiple ADC channels are read in a single sketch

paulvha, 2020-02-20:
In correct ADC readings when multiple ADC channels are read in a single
sketch. Sometimes a single ADC reading will fail as well.

Reproduce

In a single sketch read in the internal battery and the internal temperature
When FIRST reading temperature and then battery it will work. However
if FIRST reading battery and temperature, it will fail.

Root cause
ADC is enabled too early while not all the changes have been made.

From the datasheet page 825 :
This bit enables the ADC module. While the ADC is enabled, the ADCCFG
and SLOT Configuration register settings must remain stable and
unchanged. All configuration register settings, slot configuration settings
and window comparison settings should be written prior to setting the
ADCEN bit to '1'.

Solution :

Do not enable ADC in the (one time) setup, but make sure it is disabled
when selecting a (new) channel and enable only after this last change.

###########  in ap3_analog.cpp  ################

*****  routine  ap3_adc_setup() *********
REMOVE enable at the end

    // Enable the ADC.
    if (AM_HAL_STATUS_SUCCESS != am_hal_adc_enable(g_ADCHandle))
    {
        //Serial.println("Error - enabling ADC failed.\n");
        return AP3_ERR;
    }

****** routine ap3_change_channel(uint8_t padNumber) *******

ADD in the top :

    if (AM_HAL_STATUS_SUCCESS != am_hal_adc_disable(g_ADCHandle))
    {
        //Serial.println("Error - disabling ADC failed.\n");
        return AP3_ERR;
    }

ADD at the end (above return AP3_OK;)

    if (AM_HAL_STATUS_SUCCESS != am_hal_adc_enable(g_ADCHandle))
    {
        //Serial.println("Error - enabling ADC failed.\n");
        return AP3_ERR;
    }

oclyke, 2020-02-27:
@paulvha I created a PR with your suggested changes. I will request a review from the engineer who is more familiar with the ADC. Thanks for your contributions!

paulvha, 2020-02-27:
Thanks. make sure to include the other remark about the ADC sometime providing old readings.
Regards,

closed 2020-05-05 by oclyke