sparkfun/Arduino_Apollo3 #132: The first Serial.print() in a sketch is displaying garbage

paulvha, 2020-02-20:
The first Serial.print() in a sketch is displaying garbage

Reproduce
Create a sketch, in setup

Serial.begin(115200);
Serial.println(Test1);
Serial.println(Test2);

Compile and load and start. The Test1 will be displayed as with garbage
or as Garbage. Test2 will show normal

Root cause
First initialize the UART, then intialise the pin

Solution
Change the UART initialization  before the PIN setup

###########  in ap3_uart.cpp  ################

**************routine  _begin(void) ******************
after the testing for a valid pin move :

    //User may want to change settings mid-sketch. Only init UART if it's new.
    if (_handle == NULL)
    {
        // Now that pins are initialized start the actual driver
        retval = (ap3_err_t)am_hal_uart_initialize(_instance, &_handle);
        if (retval != AP3_OK)
        {
            return ap3_return(retval);
        }
    }

    retval = (ap3_err_t)am_hal_uart_power_control(_handle, AM_HAL_SYSCTRL_WAKE, false);
    if (retval != AP3_OK)
    {
        return ap3_return(retval);
    }

    retval = (ap3_err_t)am_hal_uart_configure(_handle, &_config);
    if (retval != AP3_OK)
    {
        return ap3_return(retval);
    }

    UARTn(_instance)->LCRH_b.FEN = 0; // Disable that pesky FIFO

    ********************** NOW SET THE PINS ************************

    ********************** THEN ************************************

     // Enable TX and RX interrupts
    NVIC_EnableIRQ((IRQn_Type)(UART0_IRQn + _instance));
    am_hal_uart_interrupt_enable(_handle, (AM_HAL_UART_INT_RX | AM_HAL_UART_INT_TX));
    am_hal_interrupt_master_enable();

    // Service interrupts to determine idle state
    am_hal_uart_interrupt_service(_handle, 0, (uint32_t *)&_tx_idle);

    // Register the class into the local list
    ap3_uart_handles[_instance] = this;

    return retval;

nseidle, 2020-02-20:
Hi @paulvha - Thanks for this! I too had seen the problem in a few sketches but hadn't had time to dig into the root cause. We'll get this implemented shortly. Thank you for reporting the problem and a solution. 

In the future, and only if you want gold stars and piles of appreciation from us, a PR would be ideal.

paulvha, 2020-02-20:
Not for the gold stars etc.. but just to make it simple, I will consider a pull request next time !

oclyke, 2020-02-27:
@paulvha is the above PR (#134) what you mean?

paulvha, 2020-02-27:
hi 
Yes that is it. I see a remark form Nathan that he is still having issues sometimes, mine have completely gone after this change. (and I do lot of compiles and test on Edge and Redboard ATP) That said, as I diagnosed the issue earlier, I could also reduce the problem by adding a delay 10ms after setting the pins and before enabling UART. So maybe by first setting the UART and then enabling the pins, before enabling interrupts, adding 10ms might help even more. I got 'ms' definition from the other Apollo3 code and just added on top in the file. Then I just did ms (10) :
#define delay_ms(ms)                                                          \
    if (1)                                                                    \
    {                                                                         \
        am_hal_clkgen_status_t sClkGenStatus;                                 \
        am_hal_clkgen_status_get(&sClkGenStatus);                             \
        am_hal_flash_delay((ms) * (sClkGenStatus.ui32SysclkFreq / 3000));     \
    }

nseidle, 2020-04-29:
I'm closing out. I haven't seen any stray print issues in a few weeks.

closed 2020-04-29 by nseidle