sparkfun/Arduino_Apollo3 #140: RTC bogus values

Colin-at-511100471793, 2020-03-20:
Hi all, I've seen some wierd values from the RTC module on this. At the beggining of the year my devices weren't happy because they thought it was month 17. I masked out some bit in the month column and they seemed happy. Then a few days ago I got some devices back from the field and they have a bunch of data logged on February 30th. I got a kick out of that :) I'm not sure what else to expect in the future so I'm switching to a software-tracked epoch time(based on 32khz), which is fine and simpler in fact. 

I don't think it's a problem w the library as it really just reads from the rtc registers, I think its on the silicon. I just wanted people to know about this silliness so they don't get any surprises later on :). 

I guess to recreate it you could seed the rtc with epochs corresponding to those days and see what it reads.

stephenf7072, 2020-03-23:
There was a bug with RTC month references changing from 0-11 to 12 (I experienced a similar kick of data logged on 30th Feb).  It's now been fixed, so update the core, re-flash your code and you should be right.  There's a new example to test for it too.

Details in the issue I raised here:
[https://github.com/sparkfun/Arduino_Apollo3/issues/118](url)

Colin-at-511100471793, 2020-03-24:
thanks stephen, sorry I didn't see that, I'm using a version thats at least a couple months old.

closed 2020-03-24 by Colin-at-511100471793