sparkfun/Arduino_Apollo3 #141: PDM/MIC config missing [SparkFun Edge]

metanav, 2020-03-22:
### Subject of the issue
I do not see MIC CLOCK and PDM related config in the the variants/edge/config/variant.h and  variants/edge/bsp/am_bsp_pins.h.

### Your workbench
* What platform are you using?
Arduino IDE 1.8.12, MacOS 10.15.3
* What version of the device are you using? Is there a firmware version?
SparkFun Edge, SparkFun Apollo3 Boards version 1.0.30 

### Steps to reproduce
The example PDM > Example1_MicrophoneOutput  uses the default config from the variant which is missing and the output is incorrect.

oclyke, 2020-03-22:
@metanav the Edge board does not have a default PDM microphone configuration because it does not have an onboard PDM microphone. The two microphones on the Edge are analog and are sampled by the Apollo3 ADC. To use PDM on the Edge you would have to find a set of compatible pins that you could solder to and attach an external PDM mic. This may be easier said than done. There are many pins available on/through the camera connector - inconvenient though they may be.

metanav, 2020-03-24:
@oclyke Thanks for letting me know. I just need to read Sparkfun Edge mic input so I guess analogRead(29) and analogRead(11) would be suffice or do I need to do some configuration?

oclyke, 2020-03-26:
Hmmm good question - I guess we have never really used the Edge microphones from Arduino. Just an oversight. 

Until now the Edge microphones have only ever (AFAIK) been used from the Ambiq HAL functions. There are examples in the **OLD** BSP repo.

[Old Edge Test Code](https://github.com/sparkfun/SparkFun_Edge_BSP/tree/master/examples/example1_edge_test)
[All Old Edge Examples](https://github.com/sparkfun/SparkFun_Edge_BSP/tree/master/examples)

I suppose this means we should really add microphone examples for the Edge board - both to the [new BSPs](https://github.com/sparkfun/SparkFun_Apollo3_AmbiqSuite_BSPs) and to this Arduino core. I'll rename this issue and reopen it to serve as a to-do

Wenn0101, 2020-10-12:
Closing due to inactivity

closed 2020-10-12 by Wenn0101