sparkfun/Arduino_Apollo3 #143: AnalogRead() will not return with incorrect pin

paulvha, 2020-03-24:
### Subject of the issue
When performing a Analogread() with a pin number that does not have analog support
the Analogread() will not return and loop forever.

### Steps to reproduce
loop() 
{
  Serial.print("test :");
  uint16_t tempadc = analogRead(37);
  Serial.print("analog pin :");
  Serial.println(tempadc);
}
The text 'test :' will be displayed, but 'analog pin :' will not be displayed.

### Root cause:
In ap3_analog.h the wrong number of entries is indicated : AP3_ANALOG_PADS where it should be
AP3_ANALOG_CHANNELS.    AP3_ANALOG_PADS = 10 where AP3_ANALOG_CHANNELS = 15,

In ap3_analog_structures.c the structures are defined as :

const ap3_analog_channel_map_elem_t ap3_analog_channel_map[AP3_ANALOG_CHANNELS] = {
   and
ap3_analog_configure_map_elem_t ap3_analog_configure_map[AP3_ANALOG_CHANNELS] = {

In ap3_analog.h the structures are referenced as :

extern const ap3_analog_channel_map_elem_t ap3_analog_channel_map[AP3_ANALOG_PADS];
   and 
extern ap3_analog_configure_map_elem_t ap3_analog_configure_map[AP3_ANALOG_PADS];

### Solution :
correctly reference the structures in ap3_analog.h as a size AP3_ANALOG_CHANNELS :

extern const ap3_analog_channel_map_elem_t ap3_analog_channel_map[AP3_ANALOG_CHANNELS];
extern ap3_analog_configure_map_elem_t ap3_analog_configure_map[AP3_ANALOG_CHANNELS];

oclyke, 2020-03-26:
@paulvha I am looking into this. I can see the mismatch in array sizes between ```ap3_analog.h``` and ```ap3_analog_structures.c```. I was unable to demonstrate an infinite while loop with any board definition. However I did see unexpected / incorrect behavior. For example in ```ap3_analog.c``` in the loop:

```
    for (indi = 0; indi < AP3_ANALOG_CHANNELS; indi++)
    {
        Serial.printf("indi: %d \n", indi);
        if (ap3_analog_configure_map[indi].pad == padNumber)
        {
            if (ap3_analog_configure_map[indi].isAnalog == false)
            {
                if (ap3_set_pin_to_analog(pinNumber) != AP3_OK)
                {
                    //Serial.println("Error - set pin to analog");
                    return 0; //Error
                }
                ap3_analog_configure_map[indi].isAnalog = true;
            }
            break;
        }
    }
    if (indi == AP3_ANALOG_CHANNELS)
    {
        Serial.println("Error - pin set to analog not found");
        return 0; //Error
    }
	Serial.printf("padNumber: %d, indi: %d --- ", padNumber, indi);
```

would run until reaching the final line and showing ```padNumber: 37, indi: 154327``` on the serial monitor. It appears that if ```indi``` is being incremented beyond AP3_ANALOG_CHANNELS then on my test setup the corresponding value for ```ap3_analog_configure_map[154327].pad``` does in fact evaluate to '37' which saves me from an infinite loop. 

The behavior of this loop is quite concerning. From what I know this loop should not exceed the expanded value of ```AP3_ANALOG_CHANNELS``` which should be ```15```. Replacing the macro with the literal value ```15``` does not resolve the issue, so that seems to eliminate macro expansion errors. 

Fixing the ```extern``` declarations in ```ap3_analog.h``` **does in fact** remedy this situation... I will make a commit to that effect but I am unhappy with my lack of understanding of what is going on. 

Even if the extern declaration shows the wrong size how would that impact the number of times that the loop is executed? That should be dependent only on the local index variable ```indi``` and the expansion of ```AP3_ANALOG_CHANNELS```. 

@paulvha can you explain this? I am eager to understand.
@nseidle any thoughts?

paulvha, 2020-03-26:
hi Owen

Looks like you waited MUCH longer than I did, but I had the same observation with 'indi' variable exceeding the maximum expected 15 and kept on counting. When I did set the maximum on 10 (instead of AP3_ANALOG_CHANNELS) it worked.  That was before making change to ap3_analog.h and probably because AP3_ANALOG_PADS = 10.

During debugging I had different code setup's and in one of them the compiler warned that trying this lookup it will result in "unexpected program behaviour". That warning was not shown when using the code as it is.  I expect to fully understand that, a deep study on the compiler optimizer logic will be needed, but that is above my paygrade 🙂

regards,
Paul

oclyke, 2020-03-26:
If you omit the serial statement within the loop it executes very quickly and still shows why the statement ```if (indi == AP3_ANALOG_CHANNELS)``` is not triggered. That's what clued me into the fact that ```indi``` was incrementing so far! Yikes 

Fair enough regarding the compiler's magic. Suffice it to say that if you do mean things to it the compiler just might bite back haha. 

In plain english: since ```ap3_analog.cpp``` uses the declaration of ```ap3_analog_configure_map``` from ```ap3_analog.h``` which it *_says_ said* has only ```AP3_ANALOG_PADS (10)``` elements trying to access the 11th entry is technically ill-defined. Anything could happen! (I would like to read that portion of the C / C++ specification but will leave that for a lazy afternoon sometime)

I'll turn my branch into a PR and get it in the next release. Thanks a bunch for pointing this out!

PR #145 (2020-03-26): fix extern declaration of analog channel and config maps
Fixes #143 but can't explain precisely why - probably compiler specification magic. Not to mention that there was an honest error in the codebase.
commit cafc527e (2020-03-26): fix extern declaration of analog channel and config maps

fixes #143 but can't explain exactly why
closed 2020-03-26 by oclyke