sparkfun/Arduino_Apollo3 #147: Weird Initial Gain in PDM Audio

justiceamoh, 2020-04-08:
### Subject of the issue
When acquiring audio from the PDM microphone, the gain (amplitude level) of the audio is fairly high for the first few seconds, but drops significantly afterwards. See sample audio clips and their waveforms showing this issue:

[image]

[recording_0408_1536.wav.zip](https://github.com/sparkfun/Arduino_Apollo3/files/4452632/recording_0408_1536.wav.zip)

[image]

I see this same issue even outside Arduino when I use the AmbiqSDK and Sparkfun Makefiles. This behavior can be explained by an Automatic Gain Control (AGC) but I've looked and I don't think the [microphone](https://media.digikey.com/pdf/Data%20Sheets/Knowles%20Acoustics%20PDFs/SPH0641LM4H-1.pdf) has an AGC. I've also tried different gain levels in the PDM interface and that makes no difference. The PDM configuration also has a softmute variable but changing that has no effect either. 

Any ideas what could be causing this and how to turn it off? 

### Your workbench
- macOS Catalina
- Sparkfun Redboard Artemis Nano
- Arduino 1.8.10, Sparkfun Apollo3 Boards 1.0.28

### Steps to reproduce
To reproduce, run the PDM example at: 
`Files -> Examples -> PDM -> Example4_RecordToWav`, with the corresponding [python script here](https://gist.github.com/justiceamoh/a31a93cacd6ba2c90df5bb4fd65cf946).

Note: you should reset the board just before you run the script since the issue occurs in the first 5-10 seconds.

oclyke, 2020-04-13:
Nate, I assigned you since you are much more familiar with PDM than me. Let me know if you want me to handle it regardless.

justiceamoh, 2020-04-24:
I think I figured out the issue. First, an additional observation I made was that when I circumvented UART transmission and stored directly to an SD card, the audio was just low-amplitude throughout. 

Turns out it had to do with the Endianness of the samples as they are transmitted. When the files are loaded as **Little-Endian** 16 bit PCM samples, you see the issue I observed. However, when the same binary files are correctly loaded as **Big-Endian** PCM samples, then it conforms to expectation. See image below:

[image]

closed 2020-04-24 by justiceamoh

oclyke, 2020-05-05:
Ahh that's good to know and have documented here. Thanks Justice!