sparkfun/Arduino_Apollo3 #151: Multiple interrupts cause pinMode to (eventually) freeze

stephenf7072, 2020-04-13:
### Subject of the issue
I made the sleepWithGPIOInterrupts example and it was great, although in my application I have 3 buttons, all of which should trigger an interrupt.  Mostly though, the uC sleeps for a fixed time period and is woken by the timer.

Two instances of the interrupt code seem to be ok, but three or more will cause a hang after X number of sleeps and timer wakeups.  The hang is in pinMode, and the number of sleeps before hang varies from 145 to 1177 depending on how many GPIO interrupts there are, and whether they are attached to valid or invalid pins.  It's really weird, and I've been staring at it for days, am out of ideas for the moment.

### Your workbench
I'm using a custom Artemis board, being programmed as an ATP board, core 1.0.30 on Arduino IDE 1.8.12, Windows 10.  Don't really need the pins to trigger the interrupt, as it's the timer that is failing.

### Steps to reproduce
Here's my complete test code, including comments on test cases and results:
`
//(from 2020-02-21 Sparkfun DeepSleep_with_Work example)
//Called once number of milliseconds has passed
extern "C" void am_stimer_cmpr6_isr(void)
{
  uint32_t ui32Status = am_hal_stimer_int_status_get(false);
  if (ui32Status & AM_HAL_STIMER_INT_COMPAREG)
  {
    am_hal_stimer_int_clear(AM_HAL_STIMER_INT_COMPAREG);
  }
}

//Sleeping
#define GPIO_WAKE 1
#define TIMER_WAKE 2
#define GPIO_TIMER_WAKE 3
#define TIMER_GPIO_WAKE 3

uint32_t msToSleep = 5000;  //This is the user editable number of ms to sleep between RTC checks
#define TIMER_FREQ 32768L //Counter/Timer 6 will use the 32kHz clock
uint32_t sysTicksToSleep = msToSleep * TIMER_FREQ / 1000;
bool awakeFlag = true;  //Stops wakeFromSleep() being run if the system is already awake

//Misc
#define interruptPin1 35
#define interruptPin2 19
#define interruptPin3 18  //-1
#define interruptPin4 17  //-1

//With all as -1, doesn't freeze

//Only pin 35 (as interruptPin1), doesn't freeze
//Only pin 19 (as interruptPin1), doesn't freeze
//Only pin 19 (as interruptPin2), freezes at loop 801
//Only pin 18 (as interruptPin3), freezes at loop 801
//Only pin 17 (as interruptPin4), freezes at loop 801

//Pin 35 & 19 (as interrupt pins 1 & 2), doesn't freeze
//  Removed pin 3/4 calls to pinMode, doesn't freeze
//  Removed pin 3/4 calls to detachInterrupt after sleeping, doesn't freeze
//  Removed pin 3/4 calls to attachIinterrupt, freezes at loop 1177
//Pin 35 & 19 (as interrupt pins 1 & 3), doesn't freeze
//Pin 35 & 17 (as interrupt pins 1 & 4), doesn't freeze
//Pin 18 & 17 (as interrupt pins 3 & 4), freezes at loop 265
//Pin 19 & 18 (as interrupt pins 2 & 3), freezes at loop 265

//Pin 35,19,18 (as interrupt pins 1,2,3), freezes at loop 147
//Pin 35,18,17 (as interrupt pins 1,3,4), freezes at loop 147

//With all enabled, freezes at loop 147

//Commented out the #define and all traces of interruptPin3 & 4, and freezes at loop 1177
//Commented out the #define and all traces of interruptPin4, and freezes at loop 312

//With all 4 as real pins, tried moving the block of detachInterrupt calls around (ie. after processor is re-woken), didn't seem to make a difference

uint32_t loopCount = 1;
bool DEBUG = false;

void setup()
[... 168 lines cut ...]
  recoverFromDeepSleep(); //Wake the processor, but don't do anything fancy like initialising Serial or whatever
//  am_hal_stimer_compare_delta_set(6, 0);  //Or, force the timer to run out, and resume from where you left off.
}

// GPIO Interrupt Service Routine
void myGPIO_ISR3(void)
{
  detachInterrupt(digitalPinToInterrupt(interruptPin3));  //Stop interrupt from being triggered again
  recoverFromDeepSleep(); //Wake the processor, but don't do anything fancy like initialising Serial or whatever
//  am_hal_stimer_compare_delta_set(6, 0);  //Or, force the timer to run out, and resume from where you left off.
}

// GPIO Interrupt Service Routine
void myGPIO_ISR4(void)
{
  detachInterrupt(digitalPinToInterrupt(interruptPin4));  //Stop interrupt from being triggered again
  recoverFromDeepSleep(); //Wake the processor, but don't do anything fancy like initialising Serial or whatever
//  am_hal_stimer_compare_delta_set(6, 0);  //Or, force the timer to run out, and resume from where you left off.
}`

### Expected behaviour
The code should just keep on running forever.

### Actual behaviour
It stops at repeatable points during a call to pinMode

helgerustad, 2020-05-05:
I suspect there is a missing line: "gpio_num_isr --;" at the end of function "detachInterrupt" in the core library file "ap3_gpio.cpp". The end of the function should be:

    // Clear out the last entry
    gpio_isr_entries[gpio_num_isr].pad = 0;
    gpio_isr_entries[gpio_num_isr].callback = NULL;
    gpio_isr_entries[gpio_num_isr].mode = LOW;
    gpio_isr_entries[gpio_num_isr].arg = NULL;
    **gpio_num_isr --;** 

I have an application the attaches and detaches interrupts regularly and "gpio_num_isr" was growing, causing multiple calls to the isr routine and eventually crash. The correction seems to solve the problem for me. This could be the cause for Stephen's problem also.

stephenf7072, 2020-05-06:
@helgerustad you have made my day.  

Applying this fix (or at least testing on the current release-candidate branch which includes it [https://github.com/sparkfun/Arduino_Apollo3/tree/release-candidate](url) ) I can't make any crashes using the sample code originally posted, so I will mark the issue closed.  

I'm waiting on results from my more complex datalogging application, but would start a new issue if there are any problems.

PR #161 (2020-05-05): Fix detach interrupt
addresses comment by @helgerustad in #151 (but fairly sure this does not fix #151 directly)
closed 2020-05-06 by stephenf7072