sparkfun/Arduino_Apollo3 #167: SparkFun edge Qwiic modules not working

metanav, 2020-05-11:
### Subject of the issue
Any I2C Qwiic module is not working. It was working until last month. I checked with SparkFun humidity sensor (SHTC3) and SparkFun 9DoF IMU (ICM-20948)

### Your workbench
* What platform are you using?
MacOS
* What version of the device are you using? Is there a firmware version?
SparkFun Edge
* How is the device wired to your platform?
USB C
* How is everything being powered?
USB C

metanav, 2020-07-03:
Any updates? How can I debug it myself?

oclyke, 2020-07-06:
Hi @metanav sorry this slipped through the cracks. Are you able to use a DMM or logic analyzer to confirm that there are proper I2C signals sent out of the Qwiic connector?

prerna-khanna, 2020-07-07:
> ### Subject of the issue
> Any I2C Qwiic module is not working. It was working until last month. I checked with SparkFun humidity sensor (SHTC3) and SparkFun 9DoF IMU (ICM-20948)
> 
> ### Your workbench
> * What platform are you using?
>   MacOS
> * What version of the device are you using? Is there a firmware version?
>   SparkFun Edge
> * How is the device wired to your platform?
>   USB C
> * How is everything being powered?
>   USB C
> * Are there any additional details that may help us help you?
>   The onboard LIS2DH12 3-axis accelerometer is working which is also using I2C.
>   Also, when I connect the Qwiic module the power LED is lit on the Qwiic module so it seems not a power issue.
> 
> ### Steps to reproduce
> I just connected Qwiic module and uploaded the relevant example code.
> 
> ### Expected behaviour
> It should print data from the Qwiic module.
> 
> ### Actual behaviour
> It is always getting stuck at initializing/detecting the Qwiic module.
> 
> Is it hardware issue? Is there any way to detect / investigate the Qwiic port using code or debugger? I tried i2cdetect arduino example with no success.

I am not able to use the on board accelerometer too. It says no accelerometer detected. I am using the Sparkfun LIS2DH12 library.
Also, can we use GPIO pins as SPI/ I2C?

oclyke, 2020-07-07:
Can you please post the code you are using?

prerna-khanna, 2020-07-07:
> Can you please post the code you are using?

I am using this example code: https://github.com/sparkfun/SparkFun_LIS2DH12_Arduino_Library/blob/master/examples/Example1_BasicReadings/Example1_BasicReadings.ino

nseidle, 2020-07-07:
@prerna-khanna The accelerometer on the Edge is wired to SCL3/SDA3 so the library example will fail because it assumes you are using Wire. If you're using the library, checkout Example 6:

https://github.com/sparkfun/SparkFun_LIS2DH12_Arduino_Library/blob/master/examples/Example6_AdvancedWire/Example6_AdvancedWire.ino

In the Edge variant, WireAccel is defined to use IOM3 so you should simply we able to add this to the top of your example 1:

    WireAccel.begin();

    if (accel.begin(0x19, WireAccel) == false)
    {
    ....

prerna-khanna, 2020-07-08:
Thanks a lot, this worked!

closed 2020-10-12 by oclyke

metanav, 2021-08-17:
I still have this issue, any solutions?