sparkfun/Arduino_Apollo3 #168: WARNING Watchdog timer

Colin-at-511100471793, 2020-05-20:
not really a problem with the arduino wrapper- more of a misleading documentation issue with am_hal_wdt

looking at the current version of this which I believe uses sdk 2.2

if you look in am_hal_wdt.h then you can see that ui16ResetCount is a 16 bit counter. where the datasheet for the apollo3 it clearly states that it's an 8 bit counter.

if I set up the wdt for 7200(2 hr) seconds with the 1hz counter, I would expect it not to reset if i feed it every hour. What actually happens is that it resets every 32 seconds. that's because 7200 mod 256 is 32.. so the msb's are thrown away..

so if you use this, I think the longest wdt you can make is 16*255=4080 seconds- a bit over an hour.

oclyke, 2020-05-20:
Thanks @Colin-at-511100471793! This is great feedback. Any chance you would be willing to submit this issue on our copy of the Ambiq HAL? We are trying to document stuff like this in one consistent place:

https://github.com/sparkfun/AmbiqSuiteSDK/issues

Colin-at-511100471793, 2020-05-22:
will do- @oclyke one question- does this mean that the sdk presented here is/could be different than the one ambique provides?

oclyke, 2020-05-26:
Yes - that is possible. We have had to make a handful of small patches (for example SPI full duplex functionality) in the Arduino core version. Since that could end up being a nightmare we decided to make a strong effort to document our changes - hence the other repo. Plus I've had a little interest from people at Ambiq saying that they would check in on the feedback that they can glean from the issued on that repo.

oclyke, 2020-05-26:
Out of curiosity - have you tried checking a return value from your operation that has the MSB dropped? I wonder if the HAL would warn you about that. I ask because Ambiq generally seems to use 32 bit variables for nearly everything. I find their use of a 16 bit variable odd, and would really be shocked to see an 8-bit variable in their source code.

Colin-at-511100471793, 2020-06-01:
@oclyke that's good to know, I didn't know there were big issues (like the spi thing) w the sdk, and also thx for trying to reconcile all these w ambique. 

as for the return values, I'm not sure what you mean, the am_hal_wdt_init fxn returns void so there's no success/failure return.

adamgarbo, 2020-06-01:
@Colin-at-511100471793 

I agree that it's a bit strange that Ambiq is using a 16-bit variable for an 8-bit register. Especially since it's so clearly illustrated in the datasheet. I expect it has something to do with how they shift bits around using _VAL2FLD.

[image]

I'm not sure that it's worth modifying the am_hal_wdt_config_t structure in am_hal_wdt.h and the am_hal_wdt_init() function in am_hal_wdt.c. It may be easier to advise users to use the 1/16 Hz LFRC clock and a counter in the am_watchdog_isr() similar to the watchdog.c example to achieve any length of WDT required.

Once SparkFun has a WDT library, users won't have to worry about the WDT HAL.

```
// Interrupt handler for the watchdog.
extern "C" void am_watchdog_isr(void)
{
  // Clear the watchdog interrupt.
  am_hal_wdt_int_clear();

  // Catch the first 998 watchdog interrupts, but let the 999th through untouched.
  if ( g_ui16NumWatchdogInterrupts < 999)
  {
    // Restart the watchdog.
    am_hal_wdt_restart();
  }

  // Increment the number of watchdog interrupts.
  g_ui16NumWatchdogInterrupts++;
}
```

Cheers,
Adam

adamgarbo, 2020-06-01:
Out of curiousity, I enabled the 1/16 Hz LFRC clock and timed each watchdog interrupt. I expected a period of 16 seconds but observed the watchdog trigger every ~48 seconds instead. Not sure why this is! If this holds constant, the longest WDT timer would be approximately 3 hours and 24 minutes (`.ui16InterruptCount = 255`).

```
19:41:53.536 -> Watchdog Timer Example
19:42:49.375 -> Interrupt #1, 55895 ms
19:43:39.403 -> Interrupt #2, 48109 ms
19:44:35.099 -> Interrupt #3, 48105 ms
19:45:18.880 -> Interrupt #4, 48100 ms
19:46:01.631 -> Interrupt #5, 48118 ms
```

```
am_hal_wdt_config_t g_sWatchdogConfig =
{
  // Select the Apollo Clock Rate.
  .ui32Config =
  AM_HAL_WDT_LFRC_CLK_1_16HZ |
  AM_HAL_WDT_ENABLE_RESET |
  AM_HAL_WDT_ENABLE_INTERRUPT,

  // Set WDT interrupt timeout.
  .ui16InterruptCount = 1, // 0b00000001

  // Set WDT reset timeout.
  .ui16ResetCount = 255
};
```

Colin-at-511100471793, 2020-06-04:
hmm, I"m not sure what to say here.. there's clearly something funky going on here.. one difference here is that I was using the 32khz clock whereas you are using the lfrc clock... maybe the AM_HAL_WDT_LFRC_CLK_1_16HZ macro is wrong..?? or maybe switched w the 32khz one??

Colin-at-511100471793, 2020-06-04:
I take that back, I'm using the lfrc as well..

stephenf7072, 2020-07-24:
I've just made pull request #237 which involves an example for the watchdog timer.  This issue is noted in the comments.

oclyke, 2020-10-12:
Closing b/c the WDT library abstracts this issue from Arduino users

closed 2020-10-12 by oclyke