sparkfun/Arduino_Apollo3 #171: Request Support for Addressable LED Libraries

oclyke, 2020-06-03:
We want to support both FastLED and NeoPixel libraries

Requests:
- [NeoPixel Request](https://forum.sparkfun.com/viewtopic.php?f=171&t=53049)

Resources:
- [NeoPixel PR](https://github.com/adafruit/Adafruit_NeoPixel/pull/205)

PaulZC, 2020-06-03:
Hi Owen (@oclyke),
FastLED should work OK if you pull the latest code from the repo. I added Artemis support a few weeks ago and my PR got merged last week. It hasn't been included in a release yet though.
https://github.com/FastLED/FastLED/pull/993
I hadn't seen the NeoPixel PR when I wrote it so apologies in advance for any possible duplication issues!
All the best,
Paul

oclyke, 2020-06-03:
Oh cool!!!!! I did not know you had been working on that. Thanks a bunch!

skyfly200, 2020-09-27:
I am getting the error "Unrecognised APOLLO3 board!"
I tried to export binary for the new Dev Kit and compile for the Nano, both gave the same error.

Looks like there is a mismatch of board definitions with those expected in the FastLED library file fastpin_apollo3.h

I noticed that the DevKit is not actualy defined in that file so I defined that, trying a format found in the board definitions for apollo3. Here is that code

```
#ifndef __INC_FASTPIN_APOLLO3_H
#define __INC_FASTPIN_APOLLO3_H

FASTLED_NAMESPACE_BEGIN

#if defined(FASTLED_FORCE_SOFTWARE_PINS)
#warning "Software pin support forced, pin access will be slightly slower."
#define NO_HARDWARE_PIN_SUPPORT
#undef HAS_HARDWARE_PIN_SUPPORT

#else

template<uint8_t PIN, uint8_t PAD> class _APOLLO3PIN {
public:
    typedef volatile uint32_t * port_ptr_t;
    typedef uint32_t port_t;

    inline static void setOutput() { pinMode(PIN, OUTPUT); am_hal_gpio_fastgpio_enable(PAD); }
    inline static void setInput() { am_hal_gpio_fastgpio_disable(PAD); pinMode(PIN, INPUT); }

    inline static void hi() __attribute__ ((always_inline)) { am_hal_gpio_fastgpio_set(PAD); }
    inline static void lo() __attribute__ ((always_inline)) { am_hal_gpio_fastgpio_clr(PAD); }
    inline static void set(register port_t val) __attribute__ ((always_inline)) { if(val) { am_hal_gpio_fastgpio_set(PAD); } else { am_hal_gpio_fastgpio_clr(PAD); } }

    inline static void strobe() __attribute__ ((always_inline)) { toggle(); toggle(); }

    inline static void toggle() __attribute__ ((always_inline)) { if( am_hal_gpio_fastgpio_read(PAD)) { lo(); } else { hi(); } }

    inline static void hi(register port_ptr_t port) __attribute__ ((always_inline)) { hi(); }
    inline static void lo(register port_ptr_t port) __attribute__ ((always_inline)) { lo(); }
    inline static void fastset(register port_ptr_t port, register port_t val) __attribute__ ((always_inline)) { set(val); }

    inline static port_t hival() __attribute__ ((always_inline)) { return 0; }
    inline static port_t loval() __attribute__ ((always_inline)) { return 0; }
    inline static port_ptr_t port() __attribute__ ((always_inline)) { return NULL; }
    inline static port_t mask() __attribute__ ((always_inline)) { return 0; }
};

// For the Apollo3 we need to define both the pin number and the associated pad
// to avoid having to use ap3_gpio_pin2pad for fastgpio (which would slow things down)
#define _FL_DEFPIN(PIN, PAD) template<> class FastPin<PIN> : public _APOLLO3PIN<PIN, PAD> {};

// Actual (pin, pad) definitions
#if defined(ARDUINO_SFE_EDGE)

#define MAX_PIN 49
_FL_DEFPIN(0, 0); _FL_DEFPIN(1, 1); _FL_DEFPIN(3, 3); _FL_DEFPIN(4, 4);
_FL_DEFPIN(5, 5); _FL_DEFPIN(6, 6); _FL_DEFPIN(7, 7); _FL_DEFPIN(8, 8); _FL_DEFPIN(9, 9);
_FL_DEFPIN(10, 10); _FL_DEFPIN(11, 11); _FL_DEFPIN(12, 12); _FL_DEFPIN(13, 13); _FL_DEFPIN(14, 14);
_FL_DEFPIN(15, 15); _FL_DEFPIN(17, 17);
_FL_DEFPIN(20, 20); _FL_DEFPIN(21, 21); _FL_DEFPIN(22, 22); _FL_DEFPIN(23, 23); _FL_DEFPIN(24, 24);
_FL_DEFPIN(25, 25); _FL_DEFPIN(26, 26); _FL_DEFPIN(27, 27); _FL_DEFPIN(28, 28); _FL_DEFPIN(29, 29);
_FL_DEFPIN(33, 33);
_FL_DEFPIN(36, 36); _FL_DEFPIN(37, 37); _FL_DEFPIN(38, 38); _FL_DEFPIN(39, 39);
_FL_DEFPIN(40, 40); _FL_DEFPIN(42, 42); _FL_DEFPIN(43, 43); _FL_DEFPIN(44, 44);
_FL_DEFPIN(46, 46); _FL_DEFPIN(47, 47); _FL_DEFPIN(48, 48); _FL_DEFPIN(49, 49);

#define HAS_HARDWARE_PIN_SUPPORT 1

#elif defined(ARDUINO_SFE_EDGE2)

#define MAX_PIN 49
[... 67 lines cut ...]
#define MAX_PIN 49
_FL_DEFPIN(0, 0); _FL_DEFPIN(1, 1); _FL_DEFPIN(2, 2); _FL_DEFPIN(3, 3); _FL_DEFPIN(4, 4);
_FL_DEFPIN(5, 5); _FL_DEFPIN(6, 6); _FL_DEFPIN(7, 7); _FL_DEFPIN(8, 8); _FL_DEFPIN(9, 9);
_FL_DEFPIN(10, 10); _FL_DEFPIN(11, 11); _FL_DEFPIN(12, 12); _FL_DEFPIN(13, 13); _FL_DEFPIN(14, 14);
_FL_DEFPIN(15, 15); _FL_DEFPIN(16, 16); _FL_DEFPIN(17, 17); _FL_DEFPIN(18, 18); _FL_DEFPIN(19, 19);
_FL_DEFPIN(20, 20); _FL_DEFPIN(21, 21); _FL_DEFPIN(22, 22); _FL_DEFPIN(23, 23); _FL_DEFPIN(24, 24);
_FL_DEFPIN(25, 25); _FL_DEFPIN(26, 26); _FL_DEFPIN(27, 27); _FL_DEFPIN(28, 28); _FL_DEFPIN(29, 29);
_FL_DEFPIN(31, 31); _FL_DEFPIN(32, 32); _FL_DEFPIN(33, 33); _FL_DEFPIN(34, 34);
_FL_DEFPIN(35, 35); _FL_DEFPIN(36, 36); _FL_DEFPIN(37, 37); _FL_DEFPIN(38, 38); _FL_DEFPIN(39, 39);
_FL_DEFPIN(40, 40); _FL_DEFPIN(41, 41); _FL_DEFPIN(42, 42); _FL_DEFPIN(43, 43); _FL_DEFPIN(44, 44);
_FL_DEFPIN(45, 45); _FL_DEFPIN(47, 47); _FL_DEFPIN(48, 48); _FL_DEFPIN(49, 49);
#define HAS_HARDWARE_PIN_SUPPORT 1

#else

#error "Unrecognised APOLLO3 board!"

#endif

#endif // FASTLED_FORCE_SOFTWARE_PINS

FASTLED_NAMESPACE_END

#endif // __INC_FASTPIN_AVR_H
```

skyfly200, 2020-09-27:
Rolling back to v1.2.1 I can compile FastLED sketches. So it must just be a rename of the board defs in the new v2 core

skyfly200, 2020-09-27:
Nope wait, I just had the board set to Yun accidentally

skyfly200, 2020-09-27:
Ok, it's working with the nano now but not the dev kit. It must be that I have the board def for it wrong in that file.

skyfly200, 2020-09-27:
Ok finally got the board def name right and it compiled. Not sure if the pins are right here still as I just copied the ATP pins above.

```
#ifndef __INC_FASTPIN_APOLLO3_H
#define __INC_FASTPIN_APOLLO3_H

FASTLED_NAMESPACE_BEGIN

#if defined(FASTLED_FORCE_SOFTWARE_PINS)
#warning "Software pin support forced, pin access will be slightly slower."
#define NO_HARDWARE_PIN_SUPPORT
#undef HAS_HARDWARE_PIN_SUPPORT

#else

template<uint8_t PIN, uint8_t PAD> class _APOLLO3PIN {
public:
    typedef volatile uint32_t * port_ptr_t;
    typedef uint32_t port_t;

    inline static void setOutput() { pinMode(PIN, OUTPUT); am_hal_gpio_fastgpio_enable(PAD); }
    inline static void setInput() { am_hal_gpio_fastgpio_disable(PAD); pinMode(PIN, INPUT); }

    inline static void hi() __attribute__ ((always_inline)) { am_hal_gpio_fastgpio_set(PAD); }
    inline static void lo() __attribute__ ((always_inline)) { am_hal_gpio_fastgpio_clr(PAD); }
    inline static void set(register port_t val) __attribute__ ((always_inline)) { if(val) { am_hal_gpio_fastgpio_set(PAD); } else { am_hal_gpio_fastgpio_clr(PAD); } }

    inline static void strobe() __attribute__ ((always_inline)) { toggle(); toggle(); }

    inline static void toggle() __attribute__ ((always_inline)) { if( am_hal_gpio_fastgpio_read(PAD)) { lo(); } else { hi(); } }

    inline static void hi(register port_ptr_t port) __attribute__ ((always_inline)) { hi(); }
    inline static void lo(register port_ptr_t port) __attribute__ ((always_inline)) { lo(); }
    inline static void fastset(register port_ptr_t port, register port_t val) __attribute__ ((always_inline)) { set(val); }

    inline static port_t hival() __attribute__ ((always_inline)) { return 0; }
    inline static port_t loval() __attribute__ ((always_inline)) { return 0; }
    inline static port_ptr_t port() __attribute__ ((always_inline)) { return NULL; }
    inline static port_t mask() __attribute__ ((always_inline)) { return 0; }
};

// For the Apollo3 we need to define both the pin number and the associated pad
// to avoid having to use ap3_gpio_pin2pad for fastgpio (which would slow things down)
#define _FL_DEFPIN(PIN, PAD) template<> class FastPin<PIN> : public _APOLLO3PIN<PIN, PAD> {};

// Actual (pin, pad) definitions
#if defined(ARDUINO_SFE_EDGE)

#define MAX_PIN 49
_FL_DEFPIN(0, 0); _FL_DEFPIN(1, 1); _FL_DEFPIN(3, 3); _FL_DEFPIN(4, 4);
_FL_DEFPIN(5, 5); _FL_DEFPIN(6, 6); _FL_DEFPIN(7, 7); _FL_DEFPIN(8, 8); _FL_DEFPIN(9, 9);
_FL_DEFPIN(10, 10); _FL_DEFPIN(11, 11); _FL_DEFPIN(12, 12); _FL_DEFPIN(13, 13); _FL_DEFPIN(14, 14);
_FL_DEFPIN(15, 15); _FL_DEFPIN(17, 17);
_FL_DEFPIN(20, 20); _FL_DEFPIN(21, 21); _FL_DEFPIN(22, 22); _FL_DEFPIN(23, 23); _FL_DEFPIN(24, 24);
_FL_DEFPIN(25, 25); _FL_DEFPIN(26, 26); _FL_DEFPIN(27, 27); _FL_DEFPIN(28, 28); _FL_DEFPIN(29, 29);
_FL_DEFPIN(33, 33);
_FL_DEFPIN(36, 36); _FL_DEFPIN(37, 37); _FL_DEFPIN(38, 38); _FL_DEFPIN(39, 39);
_FL_DEFPIN(40, 40); _FL_DEFPIN(42, 42); _FL_DEFPIN(43, 43); _FL_DEFPIN(44, 44);
_FL_DEFPIN(46, 46); _FL_DEFPIN(47, 47); _FL_DEFPIN(48, 48); _FL_DEFPIN(49, 49);

#define HAS_HARDWARE_PIN_SUPPORT 1

#elif defined(ARDUINO_SFE_EDGE2)

#define MAX_PIN 49
_FL_DEFPIN(0, 0);
_FL_DEFPIN(5, 5); _FL_DEFPIN(6, 6); _FL_DEFPIN(7, 7); _FL_DEFPIN(8, 8); _FL_DEFPIN(9, 9);
_FL_DEFPIN(11, 11); _FL_DEFPIN(12, 12); _FL_DEFPIN(13, 13); _FL_DEFPIN(14, 14);
_FL_DEFPIN(15, 15); _FL_DEFPIN(16, 16); _FL_DEFPIN(17, 17); _FL_DEFPIN(18, 18); _FL_DEFPIN(19, 19);
_FL_DEFPIN(20, 20); _FL_DEFPIN(21, 21); _FL_DEFPIN(23, 23);
[... 63 lines cut ...]
_FL_DEFPIN(0, 0); _FL_DEFPIN(1, 1); _FL_DEFPIN(2, 2); _FL_DEFPIN(3, 3); _FL_DEFPIN(4, 4);
_FL_DEFPIN(5, 5); _FL_DEFPIN(6, 6); _FL_DEFPIN(7, 7); _FL_DEFPIN(8, 8); _FL_DEFPIN(9, 9);
_FL_DEFPIN(10, 10); _FL_DEFPIN(11, 11); _FL_DEFPIN(12, 12); _FL_DEFPIN(13, 13); _FL_DEFPIN(14, 14);
_FL_DEFPIN(15, 15); _FL_DEFPIN(16, 16); _FL_DEFPIN(17, 17); _FL_DEFPIN(18, 18); _FL_DEFPIN(19, 19);
_FL_DEFPIN(20, 20); _FL_DEFPIN(21, 21); _FL_DEFPIN(22, 22); _FL_DEFPIN(23, 23); _FL_DEFPIN(24, 24);
_FL_DEFPIN(25, 25); _FL_DEFPIN(26, 26); _FL_DEFPIN(27, 27); _FL_DEFPIN(28, 28); _FL_DEFPIN(29, 29);
_FL_DEFPIN(31, 31); _FL_DEFPIN(32, 32); _FL_DEFPIN(33, 33); _FL_DEFPIN(34, 34);
_FL_DEFPIN(35, 35); _FL_DEFPIN(36, 36); _FL_DEFPIN(37, 37); _FL_DEFPIN(38, 38); _FL_DEFPIN(39, 39);
_FL_DEFPIN(40, 40); _FL_DEFPIN(41, 41); _FL_DEFPIN(42, 42); _FL_DEFPIN(43, 43); _FL_DEFPIN(44, 44);
_FL_DEFPIN(45, 45); _FL_DEFPIN(47, 47); _FL_DEFPIN(48, 48); _FL_DEFPIN(49, 49);
#define HAS_HARDWARE_PIN_SUPPORT 1

#else

#error "Unrecognised APOLLO3 board!"

#endif

#endif // FASTLED_FORCE_SOFTWARE_PINS

FASTLED_NAMESPACE_END

#endif // __INC_FASTPIN_AVR_H

```

skyfly200, 2020-09-27:
So with that the Dev Kit will work with FastLED on the v1.2.1 core

adamgarbo, 2021-01-06:
I'm new to the FastLED library and looking to add support for the MicroMod Artemis Processor (for now to the 1.2.1 Apollo Core). Going off of @PaulZC's previous work, it appears that the `fastpin_apollo3.h` should have the following added to it:

```
#elif defined(ARDUINO_AM_AP3_SFE_ARTEMIS_MICROMOD)

#define MAX_PIN 49
_FL_DEFPIN(0, 0); _FL_DEFPIN(1, 1); _FL_DEFPIN(2, 2); _FL_DEFPIN(3, 3); _FL_DEFPIN(4, 4);
_FL_DEFPIN(5, 5); _FL_DEFPIN(6, 6); _FL_DEFPIN(7, 7); _FL_DEFPIN(8, 8); _FL_DEFPIN(9, 9);
_FL_DEFPIN(10, 10); _FL_DEFPIN(11, 11); _FL_DEFPIN(12, 12); _FL_DEFPIN(13, 13); _FL_DEFPIN(14, 14);
_FL_DEFPIN(15, 15); _FL_DEFPIN(16, 16); _FL_DEFPIN(17, 17); _FL_DEFPIN(18, 18); _FL_DEFPIN(19, 19);
_FL_DEFPIN(20, 20); _FL_DEFPIN(21, 21); _FL_DEFPIN(22, 22); _FL_DEFPIN(23, 23); _FL_DEFPIN(24, 24);
_FL_DEFPIN(25, 25); _FL_DEFPIN(26, 26); _FL_DEFPIN(27, 27); _FL_DEFPIN(28, 28); _FL_DEFPIN(29, 29);
_FL_DEFPIN(31, 31); _FL_DEFPIN(32, 32); _FL_DEFPIN(33, 33); _FL_DEFPIN(34, 34);
_FL_DEFPIN(35, 35); _FL_DEFPIN(36, 36); _FL_DEFPIN(37, 37); _FL_DEFPIN(38, 38); _FL_DEFPIN(39, 39);
_FL_DEFPIN(40, 40); _FL_DEFPIN(41, 41); _FL_DEFPIN(42, 42); _FL_DEFPIN(43, 43); _FL_DEFPIN(44, 44);
_FL_DEFPIN(45, 45); _FL_DEFPIN(47, 47); _FL_DEFPIN(48, 48); _FL_DEFPIN(49, 49);
#define HAS_HARDWARE_PIN_SUPPORT 1
```

There does appear to be a small conflict between the `G2` definition in the MicroMod Artemis board file and the FastLED library:
```
In file included from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.2.1/cores/arduino/ard_sup/Arduino.h:87,
                 from /Users/adam/Documents/Arduino/libraries/FastLED/src/led_sysdefs.h:64,
                 from /Users/adam/Documents/Arduino/libraries/FastLED/src/FastLED.h:41,
                 from /Users/adam/Documents/Arduino/libraries/FastLED/src/hsv2rgb.cpp:4:
/Users/adam/Documents/Arduino/libraries/FastLED/src/hsv2rgb.cpp: In function 'void hsv2rgb_rainbow(const CHSV&, CRGB&)':
/Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.2.1/variants/artemis_micromod/config/variant.h:96:12: error: expected unqualified-id before numeric constant
 #define G2 34
            ^~
/Users/adam/Documents/Arduino/libraries/FastLED/src/hsv2rgb.cpp:293:19: note: in expansion of macro 'G2'
     const uint8_t G2 = 0;
```

However, once `G2` is manually commented out, the FastLED Blink example code compiles and can be uploaded but the state of the LED (in this case a [SparkFun WS2812B](https://www.sparkfun.com/products/13282)) never changes.

Are any other files required to be changed?

adamgarbo, 2021-01-06:
Hmm, actually with the above modifications to `fastpin_apollo3.h` the LED does work, but only if the delays included in the code do not exceed 349 milliseconds. For example, the following code blinks the LED correctly: 

```
#include <FastLED.h>
CRGB leds[1];
void setup() { FastLED.addLeds<WS2812B, 0, GRB>(leds, 1); }
void loop() {
  leds[0] = CRGB::White; FastLED.show(); delay(349);
  leds[0] = CRGB::Black; FastLED.show(); delay(349);
}
```

This does not:
```
#include <FastLED.h>
CRGB leds[1];
void setup() { FastLED.addLeds<WS2812B, 0, GRB>(leds, 1); }
void loop() {
  leds[0] = CRGB::White; FastLED.show(); delay(350);
  leds[0] = CRGB::Black; FastLED.show(); delay(350);
}
```

The LED seems to default on and never change. Similar behaviour is experienced when trying to set and leave the colour of the LED.

Cheers,
Adam

Wenn0101, 2021-07-20:
closed due to inactivity

closed 2021-07-20 by Wenn0101