sparkfun/Arduino_Apollo3 #205: Implement freeMemory() function

oclyke, 2020-06-30:
requested on SparkFun forums
https://forum.sparkfun.com/viewtopic.php?f=170&t=53270

stephenf7072, 2020-07-23:
After a bruising experience debugging a memory leak (caused by opening SD card files and not closing them), I've written the below code.  Can't for the life of me get a meaningful stack pointer result (needs updating in 2 places), but can be used to find heap-related memory leaks.  I was going to do a pull request but figured you'll have a better idea of where to put it, and how to deal with the global variables - ie. char _end should be the last global variable declared for it to work properly.

`//Adapted by Stephen Fordyce from code at https://forum.arduino.cc/index.php?topic=285436.0
#include <malloc.h>

//extern "C" char _end;
char _end;
extern "C" char *sbrk(int i);

#define RAM_START_ADDR 0x10010000 //Decimal: 268500992, per p71 of Apollo3 datasheet
#define RAM_END_ADDR 0x1005FFFF  //Decimal: 268828671, per p71 of Apollo3 datasheet
//Apollo3 has Decimal 327679 bytes of memory between RAM_START_ADDR & RAM_END_ADDR

void printRAMUsage()
{  
  char *heapend = sbrk(0);
//  register char * stack_ptr asm( "sp" );  //THIS DIDN'T WORK EITHER
  uint32_t stack_ptr = __get_MSP();  //THIS DOESN'T WORK (fix in 2 spots)
  Serial.println("****Printing Current RAM Usage****");
  Serial.print("Total RAM on board:"); Serial.println(getTotalRAM);
  Serial.println("Addresses:");
  Serial.print("  start RAM: 0x"); Serial.println(RAM_START_ADDR,HEX);
  Serial.print("   heap end: 0x"); Serial.println((int)heapend,HEX);
  Serial.print("  stack ptr: 0x"); Serial.print("(Wrong?)"); Serial.println((uint32_t)stack_ptr,HEX);
  Serial.print("    end RAM: 0x"); Serial.println(RAM_END_ADDR,HEX);
  Serial.println("Ram used (bytes): ");
  Serial.print("  static(globals):  "); Serial.println(getGlobalsRAM);
  Serial.print("  dynamic(heap/malloc): "); Serial.println(getHeapRAM);  
  Serial.print("  stack(locals):   "); Serial.print("(Wrong?)"); Serial.println(getStaticRAM);
  Serial.print("Estimation free Ram(byte): "); Serial.print("(Wrong?)"); Serial.println(getFreeRAM);
}

uint32_t getTotalRAM()
{
  uint32_t totalRAM = (uint32_t)RAM_END_ADDR-(uint32_t)RAM_START_ADDR;  
  return totalRAM;
}

uint32_t getGlobalsRAM()
{
  uint32_t globalsRAM = (uint32_t)&_end - (uint32_t)RAM_START_ADDR;
  return globalsRAM;
}

uint32_t getHeapRAM()
{
  struct mallinfo mi = mallinfo();  //mallinfo() definition: https://man7.org/linux/man-pages/man3/mallinfo.3.html  
  uint32_t heapRAM = (uint32_t)mi.uordblks;
  return heapRAM;
}

uint32_t getStaticRAM()
{
//  register char * stack_ptr asm( "sp" );  //THIS DIDN'T WORK EITHER
  uint32_t stack_ptr = __get_MSP();  //THIS DOESN'T WORK (fix in 2 spots)
  uint32_t stackRAM = (uint32_t)RAM_END_ADDR - stack_ptr;
  return stackRAM;  
}

uint32_t getFreeRAM()
{
  return getTotalRAM - getGlobalsRAM - getHeapRAM - getStackRAM;
}`

stephenf7072, 2021-04-02:
Mentioned here (with some extra code that may be useful): [https://forum.sparkfun.com/viewtopic.php?f=169&t=55440&p=224631#p224631](url)

Wenn0101, 2021-07-20:
Closed due to inactivity. Please create a new issue if help is still needed.

closed 2021-07-20 by Wenn0101