sparkfun/Arduino_Apollo3 #238: artemis nano 2nd i2c port undefined

Colin-at-511100471793, 2020-07-27:
Hi there, I was just wondering if in the artemis nano variant.h file you could define more than one i2c port. There's one called sda3 & scl3 on the silkscreen on bottom of the board but wire1 and wire2 are clearly not defined( #define WIRE_INTERFACES_COUNT 1). I guess now I'll do this locally but it'd be great in future to get this straightened out..

thx- colin

Colin-at-511100471793, 2020-07-27:
getting back to this.. the changes necessary are in artemis nano's variants.h file...

// Wire defines
#define WIRE_INTERFACES_COUNT 2

#define WireQwiic Wire // Giving Wire an alias of "WireQwiic" in case people want to use it
#define AP3_Wire_IOM 2 // Specify that Wire uses IOMaster instance 2
#define AP3_Wire1_IOM 3   //exposes sda/scl3 for 2nd wire bus

if you could drop this in the next rev that'd be great :)

Wenn0101, 2020-10-12:
Fixed in v2.0.0

closed 2020-10-12 by Wenn0101