sparkfun/Arduino_Apollo3 #256: Artemis Thing Plus BLE

andreh1111, 2020-08-30:
Hi everyone,
I have an Artemis Thing Plus board and I need to interface it as a BLE server. Is there any example I can use as starting point for my application?

Thanks to anyone who will answer

oclyke, 2020-08-30:
Try replacing your Arduino core installation with the latest from this repo - then download and try out the ArduinoBLE library

andreh1111, 2020-09-09:
Hi @oclyke, I followed your suggestion but I had some problems compiling for Artemis Thing Plus. In the attached image you can see the compiling problem including the ArduinoBLE library. I would also like to report that the problem does not appear if the selected board is Arduino Nano 33 IoT (one of the boards this library is intended for). Do you have some suggestion to face this problem? Thank you very much.
[image]

oclyke, 2020-09-09:
By the text in the lower right of your IDE window I can see that you must still be on a v1 branch of the Arduino core. That would cause the ArduinoBLE library not to work. 

Fortunately I made the official v2 release through the Arduino Board Manager last night. Try installing v2.0.0 and then try out some of the built-in examples (they will be under ```Examples->INCOMPATIBLE->ArduinoBLE```) [don't worry they actually are compatible but Arduino thinks that they are not]

dobra-dobra, 2020-09-10:
Hi @oclyke, 1.2.0 seems to be the newest version available in Board Manager.

oclyke, 2020-09-10:
Ah! Yes good point. I apologize because I forgot to mention that we've changed the board managers JSON url that needs to be pasted into Arduino preferences. 

That file is now maintained as part of this repo. The proper link to use is updated in the Installation section of the readme. Will you give that a try and let me know how it goes?

andreh1111, 2020-09-10:
Hi @oclyke I just want to thank you very much for your amazing work and availability. Now the BLE on my Artemis Thing Plus works great :)

closed 2020-09-18 by andreh1111