sparkfun/Arduino_Apollo3 #262: Arduino functions inconsistent / not working

kerryeven, 2020-09-11:
Platform: Windows 10 19041.450 Arduino 1.8.12 Java 1.8.0_261
Device: Tried Artemis Redboard Nano and Edge2 and no board connected
Core location: c/Users/Me/AppData/Local/Arduino15/packages/SparkFun/hardware/apollo3/2.0.0
Arduino Preferences: Arduino_Apollo3 ... package_sparkfun_apollo3_index.json

Description: Have been using Apollo3 Arduino through version 1.0.30. Updated to 2.0 after adding preference above.

Very simple Arduino sketch.  Serial works on Edge2 but not Nano, LED_BUILTIN lights on Nano but not Edge 2.
* Comments show various items I tried.  Serial's commented out to get LED on for Nano.  Serial Works for Edge2, however did lock up port at one point such that edge2 acted like nano - terminal shows just fist couple of characters of Booting... (ie: BO).  Had to re-boot to clear usb port, then Edge2 worked again.  

So close to modifying ArduinoBLE example for conrolling LED from phone, but can't get LED to light on Edge2.  That would have the same functionality as the old Example8_BLE you guys had.

** Edge2 works great for ArduinoBLE Hello World Example.  Nano locks up, possibly due to Serial problem. 

Arduino Sketch:

#define BAUD 9600      // any number, common choices: 9600, 115200, 230400, 921600
#define CONFIG SERIAL_8N1 // a config value from HardwareSerial.h (defaults to SERIAL_8N1)
const PinName ledPinName = LED1; //Works for Nano. Does not work for Edge2
//const pin_size_t ledPinNumber = LED_BUILTIN; //Works for Nano. Does not work for Edge2

void setup() {
  // put your setup code here, to run once:
  //Serial.begin(BAUD);
  //Serial.println("Booting...");  //Program stops after a couple of char's on Nano, works for Edge2
  pinMode(ledPinName, OUTPUT); //LED_BUILTIN works for Nano, not for edge2, tried LED1,LED_RED,LED2
  digitalWrite(ledPinName, HIGH); //LED_BUILTIN works for Nano, not for edge2, tried LED1,LED_RED,LED2
  //pinMode(ledPinNumber, OUTPUT); //LED_BUILTIN works for Nano, not for edge2, tried LED1,LED_RED,LED2
  //digitalWrite(ledPinNumber, HIGH); //LED_BUILTIN works for Nano, not for edge2, tried LED1,LED_RED,LED2
  //Serial.println("LED should be on"); //Never gets here on Nano, Works for Edge2 

}

void loop() {
  // put your main code here, to run repeatedly:

}

Wenn0101, 2020-09-11:
Thanks for the report!
I'll look into this and see what I can find.

kerryeven, 2020-09-12:
More info:  
Sparkfun SVL uploads but does not allow LED to turn on, while ASB does work at least for Artemis Nano (SVL is still noted as recommended).

delay(...) not working.  Only have following in code for Nano in setup.  Light turns on but does not turn off. (Used HIGH and LOW as well).
  pinMode(LED_BUILTIN, OUTPUT); //LED_BUILTIN works for Nano, not for edge2, tried LED1,LED_RED,LED2
  digitalWrite(LED_BUILTIN, 1); //LED_BUILTIN works for Nano, not for edge2, tried LED1,LED_RED,LED2
  delay(5000);
  digitalWrite(LED_BUILTIN, 0);

Haven't tried analogWrite for pwm yet or digital write to a pin but????

kerryeven, 2020-09-19:
Fix for Nano not printing with Serial.print
in file: 2.01/variants/SFE_ARTEMIS_NANO/variant.cpp
  - comment out line: Serial1(SERIAL1_TX, SERIAL1_RX);

Edge2 variant.cpp does not have this line and prints fine.  115200 works better than 9600 which looses char's.

kerryeven, 2020-10-19:
1st nano still not working on 2.0.2 (does blink on 1.0.30).  Tried 2nd nano and that one WORKS!.

oclyke, 2020-10-20:
> Working Blinky but MUST CONNECT TERMITE AT LEAST 2 SECONDS AFTER UPLOAD DONE

Excellent - we have recently identified the same error mode. Here's the issue:
https://github.com/sparkfun/Arduino_Apollo3/issues/254
We will be looking into it.

Is it appropriate to close this issue in favor of the other?

kerryeven, 2020-10-20:
I would vote to leave it open as the serial for the Nano is referring to the DAPLink uart?  I needed to comment out the reference to get it to work so needs to be changed in core?
Also, I haven't checked since 2.0 but I could not get the LED's on Edge2 to light.
Thanks, K

closed 2021-05-12 by Wenn0101