sparkfun/Arduino_Apollo3 #269: ipaddress.h missing (not where it should be)

lpdunwell, 2020-09-23:
Ethernet2\src/Ethernet2.h:15:10: fatal error: IPAddress.h: No such file or directory

oclyke, 2020-09-23:
This file [IPAddress.h](https://github.com/arduino/ArduinoCore-API/blob/7f8de5869ee3de7a915bf1db64313c08595b3ac3/api/IPAddress.h) is included as a submodule from the official ArduinoCore-API repo. Can you provide more information about this error?

Perhaps the library you are using includes IPAddress.h directly rather than via the ```Arduino.h``` header file. That behaviour would not be "defined" afaik. It would be better for that library to include ```Arduino.h``` which would subsequently include ```IPAddress.h```

lpdunwell, 2020-09-24:
Test anything from the ethernet2 lib for more details.
the problem is in the artemis board package.
please just fix it.

oclyke, 2020-10-12:
The Adafruit Ethernet2 library is deprecated
https://github.com/adafruit/Ethernet2

closed 2020-10-12 by oclyke