sparkfun/Arduino_Apollo3 #270: Possible compatibility issues with the Adafruit_BusIO source files

drbrodie, 2020-09-23:
I am trying to use an Adafruit AS7341 breakout with a RedBoard Artemis.  Their sample code "read_all_channels" compiles for Arduino Uno but not for RedBoard Artemis. A post to the SparkFun forum received a response from TS-BRANDON suggested that there might be compatibility issues with the Adafruit_BusIO source files. Relevant excerpts from the error message are as attached.
[AS7341 Artemis error2.txt](https://github.com/sparkfun/Arduino_Apollo3/files/5268069/AS7341.Artemis.error2.txt)

oclyke, 2020-09-23:
```
C:\Program Files (x86)\Arduino\libraries\Adafruit_AS7341/Adafruit_AS7341.h:93: warning: "AS7341_ASTATUS" redefined

 #define AS7341_ASTATUS 0x94    ///< AS7341_ASTATUS (unused)

 

C:\Program Files (x86)\Arduino\libraries\Adafruit_AS7341/Adafruit_AS7341.h:65: note: this is the location of the previous definition

 #define AS7341_ASTATUS 0x60    ///< AS7341_ASTATUS (unused)

 

C:\Program Files (x86)\Arduino\libraries\Adafruit_AS7341/Adafruit_AS7341.h:94: warning: "AS7341_CH0_DATA_L" redefined

 #define AS7341_CH0_DATA_L 0x95 ///< ADC Channel Data

 

C:\Program Files (x86)\Arduino\libraries\Adafruit_AS7341/Adafruit_AS7341.h:66: note: this is the location of the previous definition

 #define AS7341_CH0_DATA_L 0x61 ///< AS7341_CH0_DATA_L (unused)

 

C:\Program Files (x86)\Arduino\libraries\Adafruit_AS7341/Adafruit_AS7341.h:95: warning: "AS7341_CH0_DATA_H" redefined

 #define AS7341_CH0_DATA_H 0x96 ///< ADC Channel Data

 

C:\Program Files (x86)\Arduino\libraries\Adafruit_AS7341/Adafruit_AS7341.h:67: note: this is the location of the previous definition

 #define AS7341_CH0_DATA_H 0x62 ///< AS7341_CH0_DATA_H (unused)

 
C:\Program Files (x86)\Arduino\libraries\Adafruit_AS7341\Adafruit_AS7341.h:93: warning: "AS7341_ASTATUS" redefined

 #define AS7341_ASTATUS 0x94    ///< AS7341_ASTATUS (unused)

 

C:\Program Files (x86)\Arduino\libraries\Adafruit_AS7341\Adafruit_AS7341.h:65: note: this is the location of the previous definition

 #define AS7341_ASTATUS 0x60    ///< AS7341_ASTATUS (unused)

 

C:\Program Files (x86)\Arduino\libraries\Adafruit_AS7341\Adafruit_AS7341.h:94: warning: "AS7341_CH0_DATA_L" redefined

 #define AS7341_CH0_DATA_L 0x95 ///< ADC Channel Data

 

C:\Program Files (x86)\Arduino\libraries\Adafruit_AS7341\Adafruit_AS7341.h:66: note: this is the location of the previous definition

 #define AS7341_CH0_DATA_L 0x61 ///< AS7341_CH0_DATA_L (unused)

 

C:\Program Files (x86)\Arduino\libraries\Adafruit_AS7341\Adafruit_AS7341.h:95: warning: "AS7341_CH0_DATA_H" redefined

 #define AS7341_CH0_DATA_H 0x96 ///< ADC Channel Data

 

C:\Program Files (x86)\Arduino\libraries\Adafruit_AS7341\Adafruit_AS7341.h:67: note: this is the location of the previous definition

 #define AS7341_CH0_DATA_H 0x62 ///< AS7341_CH0_DATA_H (unused)

-IC:\\Users\\Owner\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\apollo3\\1.1.1/cores/arduino/am_sdk_ap3/third_party/exactle/sw/profiles/gatt" "-IC:\\Users\\Owner\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\apollo3\\1.1.1/cores/arduino/am_sdk_ap3/third_party/exactle/sw/profiles/gap" "C:\\Program Files (x86)\\Arduino\\libraries\\Adafruit_BusIO\\Adafruit_SPIDevice.cpp" -o "C:\\Users\\Owner\\AppData\\Local\\Temp\\arduino_build_962247\\libraries\\Adafruit_BusIO\\Adafruit_SPIDevice.cpp.o"

C:\Program Files (x86)\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp: In constructor 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, int8_t, int8_t, int8_t, uint32_t, BitOrder, uint8_t)':

C:\Program Files (x86)\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:51:48: error: 'digitalPinToPort' was not declared in this scope

   csPort = (BusIO_PortReg *)portOutputRegister(digitalPinToPort(cspin));

                                                ^~~~~~~~~~~~~~~~

C:\Program Files (x86)\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:51:48: note: suggested alternative: 'digitalPinToInterrupt'

   csPort = (BusIO_PortReg *)portOutputRegister(digitalPinToPort(cspin));

                                                ^~~~~~~~~~~~~~~~

                                                digitalPinToInterrupt

C:\Program Files (x86)\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:51:29: error: 'portOutputRegister' was not declared in this scope

   csPort = (BusIO_PortReg *)portOutputRegister(digitalPinToPort(cspin));

                             ^~~~~~~~~~~~~~~~~~

C:\Program Files (x86)\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:52:15: error: 'digitalPinToBitMask' was not declared in this scope

   csPinMask = digitalPinToBitMask(cspin);

               ^~~~~~~~~~~~~~~~~~~

C:\Program Files (x86)\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:52:15: note: suggested alternative: 'digitalPinToInterrupt'

   csPinMask = digitalPinToBitMask(cspin);

               ^~~~~~~~~~~~~~~~~~~

               digitalPinToInterrupt

C:\Program Files (x86)\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:58:33: error: 'portInputRegister' was not declared in this scope

     misoPort = (BusIO_PortReg *)portInputRegister(digitalPinToPort(misopin));

                                 ^~~~~~~~~~~~~~~~~

Using library Adafruit_AS7341 at version 1.2.0 in folder: C:\Program Files (x86)\Arduino\libraries\Adafruit_AS7341 

Using library Adafruit_BusIO at version 1.5.0 in folder: C:\Program Files (x86)\Arduino\libraries\Adafruit_BusIO 

Using library Wire at version 1.0 in folder: C:\Users\Owner\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.1.1\libraries\Wire 

Using library SPI at version 1.0 in folder: C:\Users\Owner\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.1.1\libraries\SPI 

exit status 1

Error compiling for board SparkFun RedBoard Artemis.

```

These look like issues with the library. For example the first errors suggest that ```AS7341_ASTATUS``` is defined twice in the same file. Are you able to inspect the library to see why this could be true? Perhaps there are preprocessor lines that are not being satisfied when using the RedBoard Artemis

drbrodie, 2020-09-24:
Good morning Owen,
Thanks for getting back to me so quickly.  I'm afraid that inspecting the
library is well above my skill level. I don't know whether it's within the
scope of your duties but the links to the libraries are as shown below if
you should have the time to review them yourself.

https://github.com/adafruit/Adafruit_AS7341
https://github.com/adafruit/Adafruit_BusIO

Once again, thanks!
Dean

oclyke, 2020-09-24:
Happy to try to be of service! 

As usual my response generally chronicles my approach / process - and in this case it got to be a little "extended." Here's the TLDR version:

I expect that you can fix this by upgrading to the latest Apollo3 board package (```v2.x.x```). We've migrated to a new JSON URL which you will need to use to replace the existing one in Arduino preferences. Here's that link:
```https://raw.githubusercontent.com/sparkfun/Arduino_Apollo3/master/package_sparkfun_apollo3_index.json```
If for some reason you are unable to use the v2 core read on for more possible ways to fix this!

Cheers,
Owen

I just tried compiling the "read_all_channels" example for the RedBoard Artemis using the core **v2.0.1** and it was able to compile successfully. There were still warnings about re-defined preprocessor macros (such as ```AS7341_ASTATUS```) however I did not encounter the errors (such as ```error: 'portInputRegister' was not declared in this scope```)

I then reverted back to the core **v1.1.2** (which is based on the same v1 architecture as v1.1.1 which is indicated in your error messages) and was able to reproduce the errors. Doing so illuminated the issue a bit more. Admittedly I got hung up on the redefinition warnings, but what is actually stopping you from compiling is the lack of direct port manipulation functions.

This is an interesting situation. I have not (AFAIK) ever seen ```portInputRegister```, ```portInputRegister```, or ```digitalPinToBitMask``` as part of the standard Arduino API (```pinMode```, ```digitalWrite``` etc...). If they are in fact Adafruit extensions then Adafruit should know that it is "unreliable" to use them directly. There *should* be a fallback implementation that uses only standard Arduino API methods (for cross-platform compatibility).

That being said, what are the possible solutions?

1. you could upgrade to v2.0.1 - at least there those functions are somehow defined. 
  * I'll have to dig around for a sec to see if those functions actually do what they are expected to do by this library
  * (this requires using the new JSON package link: ```https://raw.githubusercontent.com/sparkfun/Arduino_Apollo3/master/package_sparkfun_apollo3_index.json``` in your Arduino preferences)
2. perhaps we could make a pull request into the Adafruit libraries that would support our boards by defining these functions when using our targets
  * that process could begin by starting an issue on their repository as well - which could refer to this issue.
3. we could consider implementing these functions in the v1 architecture (though it is not likely that this will go through -- among many other reasons it just doesn't make sense for the Apollo3 to use the concept of GPIO ports any more - all 50 of the GPIO are handled in one big 64-bit "port". There is no reason to drag that concept along with us into the future)

Upon further inspection:
* the adafruit library BusIO has preprocessor gates to switch between direct port manipulation and the Arduino standard API. (depending on whether ```BUSIO_USE_FAST_PINIO``` is defined)
* i couldn't easily tell how ```BUSIO_USE_FAST_PINIO``` is becoming defined in our v1 core (though it looks like it would have to meet one of [these criterion](https://github.com/adafruit/Adafruit_BusIO/blob/467580a5b1ff6bfc7992110e8cf404bfb78c9087/Adafruit_SPIDevice.h#L36-L55)) 
* by commenting out all those criteria i was able to force ```#undef BUSIO_USE_FAST_PINIO``` and at least get the example to compile on our v1 architecture (as expected, by using standard Arduino API calls)

Finally:
* using the v2 (mbed based) architecture I was able to confirm that ```BUSIO_USE_FAST_PINIO``` is **not** defined - this means that I expect v2 to work by using standard Arduino API calls

drbrodie, 2020-09-24:
>    1. perhaps we could make a pull request into the Adafruit libraries
>    1. we could consider implementing these functions in the v1

Hi Owen,
I really appreciate all the effort you are putting into this.
I upgraded to v2.0.1 as you suggested and it did compile but did not "run"
the sketch.  By that I mean that the only output on the serial monitor was
a debug line that I put in SETUP after *Serial.begin*. I did not get "Could
not find AS7341" which would have resulted from* !as7341.begin() *during
SETUP. I also did not get "Error reading all channels!" in the LOOP which
would have resulted from:* !as7341.readAllChannels()*. But, no channel data
printed out either, for example:
*as7341.getChannel(AS7341_CHANNEL_415nm_F1)*.
By the way, I did check that the AS7341 breakout did compile and run
successfully on an Arduino Uno so it's not a hardware issue.

So, if you want to start the process you suggested I would greatly
appreciate it.

Thanks again,
Dean

oclyke, 2020-09-29:
Hi Dean - I currently have some other issues on my plate so I can't put a full effort into this one yet. One way or another we will be around to check this out eventually, I just can't guarantee a timeframe.

If you need it to work sooner and are willing to go through the motions I can suggest what I'd usually do:
* identify a simple core functionality of the library (e.g. do a SPI transaction to read the WHOAMI register of a sensor)
* identify what steps (conceptually) are used to perform that functionality (e.g. assert the CS line, send the register address, transfer the value back by sending a dummy value, deassert CS line)
* build those steps up one bit at a time while verifying that they work as expected (use pinMode and digitalWrite to control the CS line, use the [Arduino SPI](https://www.arduino.cc/en/reference/SPI) api to do SPI transfers, use a multimeter or [logic analyzer](https://www.sparkfun.com/products/15033) to verify pin states and data transfers)
* then check to see that the library logically does the same things

Whatever you find out please don't hesitate to post here - we can offer guidance in the short term and ultimately fix any issues you may uncover.

drbrodie, 2020-09-29:
Good morning Owen,
Hey, thanks so much for keeping in touch! Though it’s definitely pushing the limits of my skill set I will take a shot at what you have suggested - maybe, at least, I will learn something along the way. I’ll let you know if I have any breakthroughs.
Dean

stephenf7072, 2020-10-01:
Looks more like a library issue to me.  Fortunately it only needs a couple of lines commented out in the library file, and both the lines and the file are mentioned in the error messages.  More details in the forum post here [https://forum.sparkfun.com/viewtopic.php?f=169&t=53898&p=219361#p219361](url)

Wenn0101, 2021-03-02:
Close due to inactivity

closed 2021-03-02 by Wenn0101