sparkfun/Arduino_Apollo3 #271: None of the Arduino SD examples compile on V2.0.1

idea--list, 2020-09-25:
Hi there!

I paired up Artemis Thing Plus board with an uSD-slot from [Polulu](https://www.pololu.com/product/2597) as it has 11 pins (needed those to play around with 4-bit SDIO).

- Using Apollo3 core V1.1.2 i can compile all the SD examples, but neither of my cards get recognized
- Using Apollo3 core V2.0.1 the examples do not even compile

I hooked up the SD-slot like that:
<table role="table">
<thead>
<tr>
<th>microSD</th>
<th>Artemis Thing Plus</th>
</tr>
</thead>
<tbody>
<tr>
<td>SCK</td>
<td>SCK (Pin 11)</td>
</tr>
<tr>
<td>DI</td>
<td>MOSI (Pin 12)</td>
</tr>
<tr>
<td>DO</td>
<td>MISO (Pin 13)</td>
</tr>
<tr>
<td>CS</td>
<td>Pin 4</td>
</tr>
<tr>
<td>VCC</td>
<td>3V3</td>
</tr>
<tr>
<td>GND</td>
<td>GND</td>
</tr>
</tbody>
</table>

I just wonder if i might have damaged the slot during soldering, or just the wires might be too long as the very same slot works when connected it to my Lolin D32 board (ESP32 has an SDIO interface enabling 4-pin communication). With that board i can read/write an old 1GB Transcend card using SD_MMC.h (both SDIO 1 and 4-pin modes work), also i can access the card using 1-bit SPI interface (however a newer SanDisk Extreme 32GB card just will not be recognized by the Lolin board in the very same setup, not even in SPI-mode).

oclyke, 2020-09-29:
Thanks @idea--list 

we won't be able to get to this issue right away, but we will move it up in priority. SparkFun has [decided to move beyond the traditional SPI signal names](https://www.sparkfun.com/spi_signal_names) so that is *part* of the issue (```MOSI```, ```MISO```, ```SCLK``` replaced by the likes of ```SDO```, ```SDI```, ```CLK```) and the other part may have to do with the library's default SD transport layer. Does it use GPIO or SPI APIs? I would have to look in more detail (because we usually use 3rd party SD libs like [SdFat](https://github.com/greiman/SdFat)) later to identify the proper solution.

If you want to take a crack at it before then I would suggest:
* checking if the library can use the Arduino SPI API (SPI, beginTransaction, transfer, endTransaction, etc...) w/o using pin names directly
* if it cannot, and it uses GPIO access (pinMode, digitalWrite) then can you "alias" the old pin names to the new pin names? (check out the [PinNames.h](https://github.com/sparkfun/mbed-os-ambiq-apollo3/blob/ambiq-apollo3-arduino/targets/TARGET_Ambiq_Micro/TARGET_Apollo3/TARGET_SFE_ARTEMIS_THING_PLUS/PinNames.h) file for the boards you are interested in for some good mbed-recognized pin names)

idea--list, 2020-09-30:
Thanks @oclyke for the response

> **Is there really anything wrong with the terms Master and Slave?**
> Yes, and it’s ok that you asked. These are terms that have centuries of history and violence packed into them. No human should be enslaved.

Holy cow! This can not be serious... i mean what do pin names do with _humans_?!
My answer: i really do not know. But hey let's just break compatibility with libraries that are not maintained on a daily level and even with code that someone might have written long ago.

For the sake of what, might i ask but there is simply no answer... Can not believe intelligent people making such a dumbass decision.

Anyway modified my variants.h file of the V2.0.1 core as you suggested, it contains now:
`#define VARIANT_SPI_SDI     SPI_SDI
#define VARIANT_SPI_SDO     SPI_SDO
#define VARIANT_SPI_CLK     SPI_CLK
#define MISO 		    SPI_SDI
#define MOSI		    SPI_SDO
#define SCK		    SPI_CLK`

This way i get this error message during compilation:
`Arduino: 1.8.12 (Windows 10), Board: "Artemis Thing Plus, 921600, Ambiq Secure Bootloader (Recommended)"
In file included from C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2Card.h:22,
                 from C:\Program Files (x86)\Arduino\libraries\SD\src/utility/SdFat.h:29,
                 from C:\Program Files (x86)\Arduino\libraries\SD\src/SD.h:20,
                 from C:\Program Files (x86)\Arduino\libraries\SD\examples\CardInfo\CardInfo.ino:24:
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:23:26: error: 'SS' was not declared in this scope
   uint8_t const SS_PIN = SS;
                          ^~
In file included from C:\Program Files (x86)\Arduino\libraries\SD\src/utility/SdFat.h:29,
                 from C:\Program Files (x86)\Arduino\libraries\SD\src/SD.h:20,
                 from C:\Program Files (x86)\Arduino\libraries\SD\examples\CardInfo\CardInfo.ino:24:
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2Card.h:64:41: error: 'SS' was not declared in this scope
     uint8_t const  SD_CHIP_SELECT_PIN = SS;
                                         ^~
exit status 1
Error compiling for board Artemis Thing Plus.
This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
` 

So next i also added `#define SS  4` to variants.h. After doing so get this message:
`C:\Program Files (x86)\Arduino\libraries\SD\src\File.cpp: In constructor 'SDLib::File::File(SdFile, const char*)':
C:\Program Files (x86)\Arduino\libraries\SD\src\File.cpp:25:37: warning: 'void* memcpy(void*, const void*, size_t)' writing to an object of type 'class SdFile' with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
     memcpy(_file, &f, sizeof(SdFile));
                                     ^
In file included from C:\Program Files (x86)\Arduino\libraries\SD\src/SD.h:20,
                 from C:\Program Files (x86)\Arduino\libraries\SD\src\File.cpp:15:
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/SdFat.h:151:7: note: 'class SdFile' declared here
 class SdFile : public Print {
       ^~~~~~
C:\Users\Peter\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.1\cores\arduino\mbed-bridge\core-implement\HardwareSerial.cpp: In member function 'int UART::printf(const char*, ...)':
C:\Users\Peter\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.1\cores\arduino\mbed-bridge\core-implement\HardwareSerial.cpp:153:19: warning: variable length array 'buf' is used [-Wvla]
     char buf[space];`

The Arduino logic is not for me... i opt out digging deeper myself

oclyke, 2020-09-30:
@idea--list the output you show after your modifications are just warnings. I've created a branch with this patch as a temporary solution.

idea--list, 2020-09-30:
@oclyke 
Thank you very much for the patch and clarification that the last ones were only warnings! Now the SD library works.
You can clearly see i am not an Arduino fanboy, i just spend as few time with it as i must to in order to be able to avoid using it... i mean if i find an arduino library, then i try to port it to mbed.

closed 2020-09-30 by idea--list