sparkfun/Arduino_Apollo3 #274: HardwareSerial operator bool not implemented

oclyke, 2020-10-02:
This [SFE forum issue](https://forum.sparkfun.com/viewtopic.php?f=172&t=53844) highlights that UART objects are sometimes used to check the status of the port via the boolean operator. 

Currently the [virtual bool operator of the HardwareSerial base class](https://github.com/arduino/ArduinoCore-API/blob/7f8de5869ee3de7a915bf1db64313c08595b3ac3/api/HardwareSerial.h#L99) is not implemented by the UART class, and so the UART class always equates to ```0``` or ```false```.

In the previous version (v1) of this core the implementation simply equated to ```1``` or ```true``` however this defeats the actual purpose of code like:
```
while(!Serial){} // wait for user to open serial monitor
```
which is meant to ensure that the user receives messages from the setup portion of the code

How to implement this functionality properly will require taking a look into the UART peripheral capabilities on the Apollo3 as well as possibly studying the implementation from other Arduino cores (for example AVR / ESP32)

paulvha, 2020-10-07:
Many / most others do as you guys do.. just return the bool-value true. Sometimes there is a check whether buffers had been assigned and thus whether begin() had been performed.  The DTR signal (remote is now connected) is triggering most of the time a reset for restart. Hence it LOOKS as if the board has been waiting... but in reality it was just reset to start again. Some boards need a bit time to stabalize the clock.. in that case add a delay before returning

One exception to this is the SAM3X processor (as in the Arduino DUE).  The device uses the USB communication device class (CDC) drivers to take advantage of the installed PC RS-232 software to talk over the USB. The CDC class is implemented in all releases of Windows ® beginning with Windows98SE. The CDC document, available at www.usb.org, describes a way to implement devices such as ISDN modems and virtual COM ports.

regards,
Paul

Wenn0101, 2020-10-12:
For now lets just have this return true.

If we decide to do something more advanced later lets open a new issue.

oclyke, 2020-10-13:
upon inspection it appears that this boolean operator is in fact implemented correctly
this suggests that the issue reported in the forum must have a different cause
closing this issue - if there is another UART / Serial related issue we can open a new thread

closed 2020-10-13 by oclyke