sparkfun/Arduino_Apollo3 #275: WDT Library Missing

oclyke, 2020-10-02:
we had it in v1...
now it's gone

adamgarbo, 2020-10-29:
Hi @oclyke 

A mystery! I'm working on the WDT library for v2.0 of the Apollo3 Core now. 

I noticed in the other libraries that you're creating an instance of the class directly from inside the library (e.g. `extern Apollo3RTC RTC;`). Is this the new standard for the Apollo3 Core?

Also, if I have `#define` statements in the `.h` file, do they fall under `# Constants (LITERAL1)` in `keywords.txt`?

Cheers,
Adam

adamgarbo, 2020-10-29:
Hey @oclyke 

I'm running into a strange issue when trying to compile the library in v2.0.0. A snippet of the error message is below:

```
/Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.0/libraries/WDT/src/WDT.cpp: In member function 'void Apollo3WDT::configure(uint8_t, uint8_t, uint8_t)':
/Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.0/libraries/WDT/src/WDT.cpp:26:12: error: base operand of '->' has non-pointer type 'Apollo3WDT'
         WDT->CFG_b.CLKSEL = clock;
            ^~
```

If I revert to creating an instance of the WDT class directly in the Arduino .ino example, the error changes to:

```
/Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.0/libraries/WDT/src/WDT.cpp: In member function 'void Apollo3WDT::configure(uint8_t, uint8_t, uint8_t)':
/Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.0/libraries/WDT/src/WDT.cpp:26:9: error: 'WDT' was not declared in this scope
         WDT->CFG_b.CLKSEL = clock;
         ^~~
```

It seems that this may be an issue with the HAL. Is the library still able to access apollo3.h where `WDT` is defined/declared? 

Cheers,
Adam

closed 2021-03-02 by Wenn0101