sparkfun/Arduino_Apollo3 #276: Wire.h fails

paulvha, 2020-10-07:
Running version 2.0.1.

Reproduce :
Make an empty sketch and just add  at the top : #include "Wire.h"
Try to compile and it fails with:
In file included from /home/paul/Arduino/sketch_oct07a/sketch_oct07a.ino:1:
/home/paul/.arduino15/packages/SparkFun/hardware/apollo3/2.0.1/libraries/Wire/src/Wire.h:11:10: fatal error: Print.h: No such file or directory
 #include "Print.h"

Root cause
Print.h is now in mbed-bridge/core-api/api directory.

Solution
Either change the line to #include "mbed-bridge/core-api/api/Print.h" or just remove #include "Print.h" from Wire.h

regards,
Paul

oclyke, 2020-10-13:
@paulvha I could not reproduce this error on the current master branch
this include directive on its own *should* indeed fail to resolve (there is no complete search path direct to ```"print.h"```), but on my system it did not
my preference would be to remove that include directive since it is already included by the ```ArduinoAPI.h``` header
opening a PR

paulvha, 2020-10-14:
I have just tried again. I am using IDE 1.8.13. 
I have removed and reinstalled the 2.0.1 library to make sure I had not changed anything and make a clean start.
Empty sketch with only #include <Wire.h> in the top. Nothing else..no code. Try to compile give the error  : 

In file included from /home/paul/Arduino/sketch_oct12a/sketch_oct12a.ino:1:
/home/paul/.arduino15/packages/SparkFun/hardware/apollo3/2.0.1/libraries/Wire/src/Wire.h:10:10: fatal error: Print.h: No such file or directory
 #include "Print.h"
          ^~~~~~~~~
Indeed it is not necessary to include the Print.h in this file, as it all compiles without any problem is commented out.

PR #282 (2020-10-16): remove Print.h include from Wire.h
fixes #276
```Print.h``` include is part of ```Arduino.h```
commit 4e7c92c9 (2020-10-16): remove Print.h include from Wire.h

fixes #276
```Print.h``` include is part of ```Arduino.h```
closed 2020-10-18 by oclyke