sparkfun/Arduino_Apollo3 #279: how to rebuild libmbed-os.a

paulvha, 2020-10-07:
Much of the kernel code (for good reason) is stored in a static library {build.variant.path}/mbed/libmbed-os.a. I am studying the library code and want to check some changes. However even if I change the source code, it will not be taken into effect as the compiler (instructed in platform.txt) is using the library. How can I rebuild the libmbed-os.a from the changed source code?

regards,
Paul

oclyke, 2020-10-07:
Great question!
Check out the CI / GitHub Action here:
[generate-variants.yaml](https://github.com/sparkfun/Arduino_Apollo3/blob/master/.github/workflows/generate-variants.yml)

That's a hint... It relies on another GitHub action to run a few arbitrary mbed build jobs. If oyu check out the arguments passed in you can see that the mbed command to build a static library is:
```mbed compile --library --source=mbed-os -m ${YOUR_TARGET_BOARD} -t ${YOUR_TOOLCHAIN}```
(your toolchain would most likely be GCC_ARM, and your target board would be one of the variant folder names e.g. SFE_ARTEMIS for the RedBoard Artemis, SFE_ARTEMIS_DK for the Artemis Dev Kit and so on)

This means you will need to get mbed installed w/ a gcc toolchain. Here's a little more info about that:
https://learn.sparkfun.com/tutorials/artemis-development-on-arm-mbed-os-betaw

paulvha, 2020-10-07:
Thanks for the information.
regards
Paul

Wenn0101, 2020-10-07:
I wanted to add. You want to use the [ambiq-apollo3-arduino](https://github.com/sparkfun/mbed-os-ambiq-apollo3/tree/ambiq-apollo3-arduino) branch when doing this. The latest version of mbed changed some namespaces and the arduinoBLE has not yet been updated to match these changes. This branch is based on an older version of mbed.

paulvha, 2020-10-16:
As I have spend good time getting this going I have documented how I had done it. Maybe useful for others as a starting point
[make_static.zip](https://github.com/sparkfun/Arduino_Apollo3/files/5392074/make_static.zip)