sparkfun/Arduino_Apollo3 #286: Lack of "Arduino.h", EEPROM library failure, HAL functions no longer recognised, general disappointment with transition to new core

stephenf7072, 2020-10-17:
Guys,
I hate to say it, but the new core continues to disappoint (and fail to compile).  I've spent a couple of hours trying out the 2.0.1 release, the SD patch version, and the release-candidate branch.

There are a whole swag of Apollo3 functions that aren't backwards compatible with previous versions.  If there was a memo detailing which ones, please make it more obvious, if there wasn't, there really should have been.  Half the examples are missing, including a couple I made the effort of contributing as pull requests in previous versions of the core.  Many of the remaining examples are oversimplified or overcomplicated.

Case in point: the I2C example is particularly hard to follow, goes off into multi-threading and other distracting stuff rather than basics, like you should replace “TwoWire” with “MbedI2C", and pay attention to the arguments, because they are different.

The lack of backwards compatibility or documentation (again, correct me if I'm wrong) is the most annoying thing, especially where everything I've come across so far could have been covered quite easily using wrapper functions or macros.  Since compilation halts at the first error (and takes about 2 minutes to get to that point), it is extremely tedious to find these by trial and error.

Regards,
Stephen

DETAILED NOTES AND ERRORS:
•	What happened to half the examples?  Including:
  o	Reset methods
  o	Low power stuff
  o	Many others

Compilation errors:
•	Using standard 2.0.1 core…
•	Compile errors:
  o	Replace “TwoWire” with “MbedI2C” for object declarations
    	Compile Error: no matching function for call to 'arduino::MbedI2C::MbedI2C(int)'
    	Original code: TwoWire myWire2(2); //Will use pads 25/27, pins labeled D6/RX1 on BlackBoard Artemis
      •	Wants pins instead of port number?
      •	MbedI2C myWire2(25,27); will at least compile
o	Replace declared RTC object with some “RTC” thing
  	Compile Error: 'APM3_RTC' does not name a type; did you mean 'AP3_ERR'?
  	CHANGE ORDER OF ARGUMENTS???
o	Replace “Uart” type with?
  	Compile Error: 'Uart' does not name a type; did you mean 'sqrt'?
  	Original code: Uart SerialExt(1, RX1Pin, TX1Pin); // Declares a Uart object called SerialExt using instance 1 of Apollo3 UART peripherals with RX on pin 43 and TX on pin 42
  	Change Uart SerialExt(1, RX1Pin, TX1Pin);
  	To UART SerialExt(TX1Pin, RX1Pin);
  	Different order, no selection of WHICH uart, different case
o	PinMode
  	Compile Error: no matching function for call to 'pinMode(int&, const am_hal_gpio_pincfg_t&)'
  	Change: pinMode(IndDetectorPin, g_AM_HAL_GPIO_OUTPUT_8);
  	Use a #define to just make it an output
  	NFI where the options are for pin drive current, etc.
o	power_adc_disable() not recognised
  	Compile error: 'power_adc_disable' was not declared in this scope
  	Try more calls to am_hal_pwrctrl_periph_disable???
o	ap3_adc_setup() not recognised
  	Replace with initializeADC();
o	'getInternalTemp' was not declared in this scope
  	Replace with getTempDegF(), getTempDegC() is soon to be available in the next core apparently

•	Library errors:
  o	(SD library) error: 'SS' was not declared in this scope
  o	(EEPROM library) error: 'FlashIAPBlockDevice' has not been declared (compilation of example fails)

•	Using SD-patched version of 2.0.1:
  o	C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.1/cores/arduino/sdk/ArduinoSDK.h:9:10: fatal error: Arduino.h: No such file or directory   	 #include "Arduino.h"

•	Using Release-Candidate (as at 2020-10-17):
  o	C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.1/cores/arduino/sdk/ArduinoSDK.h:9:10: fatal error: Arduino.h: No such file or directory   	 #include "Arduino.h"

gvansickle, 2020-10-27:
Hmmm, I was hoping against hope it was just me.  With 2.0.2 I'm still seeing at least both the no-FlashIAPBlockDevice-declaration probelm and the "No "Arduino.h" problem:

```
In file included from <command-line>:
/home/gary/.arduino15/packages/SparkFun/hardware/apollo3/2.0.2/cores/arduino/sdk/ArduinoSDK.h:9:10: fatal error: Arduino.h: No such file or directory
#include "Arduino.h"
	      ^~~~~~~~~~~
compilation terminated.
```

Any idea on workarounds?

stephenf7072, 2020-10-27:
Keep testing and reporting errors with 2.0.2, and in the meantime roll back to v1.2, that's what I'm doing.

gvansickle, 2020-10-29:
Well, I think I'm getting further with 1.2.0, but now Adafruit_BusIO is puking all over the place, starting with:

```
In file included from /home/gary/Arduino/libraries/Adafruit_DS3502/Adafruit_DS3502.cpp:37:
/home/gary/Arduino/libraries/Adafruit_DS3502/Adafruit_DS3502.h:23:10: fatal error: Adafruit_BusIO_Register.h: No such file or directory
 #include <Adafruit_BusIO_Register.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

... even though the "missing" header is in the same libraries/ dir as the DS3502 driver, and both are discovered (apparently) properly.  Then of course things go from bad to worse.

This may not be the core's fault, haven't looked real hard into it yet.

stephenf7072, 2020-10-29:
Probably not the core's fault, I've dabbled with the Adafruit_BusIO on Artemis and had to comment out a few obscure SPI things I wasn't using anyway to get it to compile and work - see this issue `https://github.com/sparkfun/Arduino_Apollo3/issues/270`

The Adafruit_BusIO might have some other supporting libraries that need installing too.

jerabaul29, 2020-12-10:
I also have the ```Arduino.h``` problem. Arduino IDE 1.8.13, Apollo library 2.0.3, all installed a few minutes ago. I get:

```
.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/cores/arduino/sdk/ArduinoSDK.h:9:10: fatal error: Arduino.h: No such file or directory
 #include "Arduino.h"
          ^~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board RedBoard Artemis.
```

jerabaul29, 2020-12-11:
I just re-installed today and the 2.0.3 works now :) . I think there was a recent push yesterday that fixed the Arduino.h problem for me :) . Many thanks! :)

Wenn0101, 2020-12-11:
@jerabaul29 I am fairly new to making the releases and tried to release 2.0.3 in the middle of the night so nobody would notice if I made a mistake. I corrected this one fairly fast, but it looks like it got you, sorry about that. 

I have been keeping this issue up mainly to track the eeprom failure and remind me of backwards compatibility problems. Are people still experiencing the Arduino.h failures? If so are you downloading thru github of the Arduino package manager? (If you are using Gmail t, make sure all your submodules are up to date, `git submodules update --init --recursive` should do the trick)

jerabaul29, 2020-12-11:
No worries at all @Wenn0101 . I think night for you is day for me ^^ :) Many thanks for fixing the release issue, looks to work fine now for compiling! :) .

Maybe this means splitting / closing and opening new issues would make things clearer to new users like me :)

stephenf7072, 2020-12-17:
Hi guys, I've had a quick go with 2.0.3 again.  Still needed to make the fixes as above, and still got stuck with the EEPROM library throwing a bunch of errors.  No issues with Arduino.h

Actually, the "Example2_Structures" example which is the only one that uses the EEPROM library doesn't even compile.

The error messages for my attempt to compile:
`In file included from sketch\A2_Inc_Glob.h:22,
                 from C:\MySketchPath.ino:1:
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h:17:51: error: expected class-name before '{' token
 class EEPROMClass : protected FlashIAPBlockDevice {
                                                   ^
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h: In member function 'void EEPROMClass::read(int, uint8_t*, uint32_t)':
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h:34:9: error: 'FlashIAPBlockDevice' has not been declared
         FlashIAPBlockDevice::read(data, idx, size);
         ^~~~~~~~~~~~~~~~~~~
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h: In member function 'uint8_t EEPROMClass::read(int)':
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h:38:14: error: invalid conversion from 'uint8_t*' {aka 'unsigned char*'} to 'int' [-fpermissive]
         read(&val, idx, 1);
              ^~~~
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h:33:19: note:   initializing argument 1 of 'void EEPROMClass::read(int, uint8_t*, uint32_t)'
     void read(int idx, uint8_t* data, uint32_t size){
               ~~~~^~~
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h:38:20: error: invalid conversion from 'int' to 'uint8_t*' {aka 'unsigned char*'} [-fpermissive]
         read(&val, idx, 1);
                    ^~~
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h:33:33: note:   initializing argument 2 of 'void EEPROMClass::read(int, uint8_t*, uint32_t)'
     void read(int idx, uint8_t* data, uint32_t size){
                        ~~~~~~~~~^~~~
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h: In member function 'void EEPROMClass::write(int, uint8_t*, uint32_t)':
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h:43:38: warning: variable length array 'scratch' is used [-Wvla]
         uint32_t scratch[scratch_size];
                                      ^
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h:44:9: error: 'FlashIAPBlockDevice' has not been declared
         FlashIAPBlockDevice::read((uint8_t*)scratch, 0, _cfg.sram_bytes);   // keep all of flash in sram in case we need to erase
         ^~~~~~~~~~~~~~~~~~~
In file included from sketch\A2_Inc_Glob.h:22,
                 from C:\MySketchPath.ino:1:
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h:47:26: error: 'FlashIAPBlockDevice' has not been declared
             int result = FlashIAPBlockDevice::program((uint8_t*)scratch, 0, 4*scratch_size);
                          ^~~~~~~~~~~~~~~~~~~
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h: In member function 'void EEPROMClass::erase()':
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h:61:38: error: 'get_erase_size' was not declared in this scope
         mbed::bd_size_t block_size = get_erase_size();
                                      ^~~~~~~~~~~~~~
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h:63:13: error: 'FlashIAPBlockDevice' has not been declared
             FlashIAPBlockDevice::erase(erased, block_size);
             ^~~~~~~~~~~~~~~~~~~
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h:65:27: error: 'size' was not declared in this scope
         } while (erased < size());
                           ^~~~
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h:65:27: note: suggested alternative:
In file included from C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3/cores/mbed-os/platform/cxxsupport/mstd_memory:38,
                 from C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3/cores/mbed-os/platform/cxxsupport/mstd_functional:31,
                 from C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3/cores/mbed-os/platform/Callback.h:27,
                 from C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3/cores/mbed-os/rtos/Thread.h:30,
                 from C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3/cores/mbed-os/rtos/rtos.h:29,
                 from C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3/cores/mbed-os/mbed.h:22,
[... cut ...]
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h:33:10: note:   conversion of argument 2 would be ill-formed:
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h:36:13: note: candidate: 'uint8_t EEPROMClass::read(int)'
     uint8_t read(int idx){
             ^~~~
C:\Users\steph\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\libraries\EEPROM\src/EEPROM.h:36:13: note:   candidate expects 1 argument, 3 provided
`

Wenn0101, 2020-12-17:
Hey @stephenf7072, thanks for the comment. The EEPROM library has almost made its way to the top of my list of tasks, but hasn't made it there yet. Probably v2.0.4. I was checking back here to comment on the Arduino.h and make sure that it was not still an issue. I'll comment back again once I have the EEPROM up and running again.

Wenn0101, 2021-03-02:
Hello, 

Many of these issues are fixed in the latest version. I am closing this, and if there are issues remaining, please open 1 issue per problem.

closed 2021-03-02 by Wenn0101

stephenf7072, 2021-03-02:
No worries, thanks.