sparkfun/Arduino_Apollo3 #293: HAL Functions No Longer Compiling and Missing Advanced Examples in V2

charlesramey, 2020-10-22:
### After upgrading to V2 of this Apollo 3 Core I was able to get Arduino BLE functions to compile but now HAL functions are not recognized
Title sort of says it all here. 

Further background info:
I am basically trying to meld the functionalities from the LowPower_WithWork example with some simple BLE functionalities. On version 1.2.1 I was able to successfully compile and run the LowPower_WithWork example but I could not import and compile the Arduino BLE library. I saw in a previous issue posted here that upgrading to V2 would fix BLE compatibility issues so I went ahead and upgraded. Now Arduino BLE does indeed compile but now the HAL functions in the LowPower_WithWork example don't compile.

### Your workbench
* I'm using the Sparkfun Apollo 3 RedBoard
* I'm programming the board from the Arduino IDE
* I'm connected to the board via a USB A to USB C cable
* Everything is being powered via the USB cable from my MacBook Air

### Steps to reproduce
This issue can be reproduced by attempting to recompile the LowPower_WithWork from the earlier V1 core after upgrading to V2 .

### Expected behaviour
Ideally these HAL functions should still work in the newer version of the core.

### Actual behaviour
Instead the HAL functions (such as ap3_adc_setup()) are unable to be compiled.

Any help would be appreciated! If y'all need help with development on this platform, I'd be happy to help out if provided with some pointers on where best to start.

charlesramey, 2020-10-22:
## Work Around
After some further digging and exploration, I realized that these ADC functions are now just defined differently. For anyone else who comes across this issue, I'd recommend using the RTC example called Example6_LowPower_Alarm as a reference. It does covers all the ADC and low power calls you'll need for sleeping the system on command.

closed 2020-10-22 by charlesramey

Marsfan, 2021-01-05:
@charlesramey Do you have any clue what happened to the advanced examples? I just noticed that the burst mode functions are also renamed.