sparkfun/Arduino_Apollo3 #294: undefined reference to `__wrap__calloc_r'

paulvha, 2020-10-25:
Trying to make a sketch with atof() fails with undefined reference to `__wrap__calloc_r'.
**reproduce**
void setup() {
 float tempAdjust = 0;
 Serial.begin(115200);
 char input[10];
 strcpy(input,"20.5");
 tempAdjust = atof(input);
 Serial.println(tempAdjust);
}

Compile fails with
/home/paul/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: /home/paul/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/softfp/libc_nano.a(lib_a-mprec.o): in function `_Balloc':
mprec.c:(.text._Balloc+0x36): undefined reference to `__wrap__calloc_r'
/home/paul/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: mprec.c:(.text._Balloc+0x48): undefined reference to `__wrap__calloc_r'
collect2: error: ld returned 1 exit status

Change the line **tempAdjust = atof(input)**; to **tempAdjust = 20**; and it will compile

**Root cause**
including atof() will trigger gcc code to allocate memory that can not be handled by the linker. 

The  issue of the '__wrap__calloc_r'  not working was known already in 2017 and fixed for export: https://github.com/ARMmbed/mbed-os/issues/3673.  BUT "mbed compile" (to create the static library) does not seem  to use the Makefile.tmpl with the proposed solution and /or actually the proposed solution is not in the Makefile.tmpl on the current used Mbed version.

Wenn0101, 2021-03-02:
is this still a problem? I just tested it on the latest version and I didn't get compilation errors?

closed 2021-03-02 by Wenn0101

paulvha, 2021-03-03:
Thanks,

Have not tried it since to reproduce and if it comes back in future project I will raise it

regards,
Paul