sparkfun/Arduino_Apollo3 #296: Dependencies Files Not Generated During Build (Arduino Build Caching)

VisualMicro, 2020-10-27:
On the Sparkfun Apollo 3 Boards (v2.0.2), there are no *.d files generated as part of the build process, which prevents caching of components during build (libraries etc).

Additional parameters can be added to the platform.txt to resolve this:
[Arduino Compiler Dependencies Caching](https://www.visualmicro.com/page/Arduino-Compiler-Dependencies-Caching.aspx)

Marsfan, 2021-01-04:
@VisualMicro Been having the same issue with v2.0.3 on Windows 10 x64. I am working on a project that has ~10 different files, and I only edit 3 of them. Compile time takes 2-3 minutes because all files are recompiled every time.

VisualMicro, 2021-01-04:
@Marsfan That is not normal, please open a thread in our forum and follow the guide in the yellow box at the top of our forum. Then we can see which IDE, which tools chains and which settings are enabled. Thanks https://www.visualmicro.com/forums/

closed 2021-01-04 by VisualMicro

VisualMicro, 2021-01-04:
Yes certainly, if you need more assiatance with this then please open a thread here https://www.visualmicro.com/forums/

Marsfan, 2021-01-04:
@VisualMicro It worked! Thank you so much!