sparkfun/Arduino_Apollo3 #298: Control of the internal LEDs not possible

mraugit, 2020-11-14:
Control of internal LEDs not possible with versions 2.0.x. With version 1.2.x the following example runs fine:
Board: Sparkfun edge

````
/*
  Blink
  Turns on and off all LEDs.

 */

// give it a name:
int led_1 = 46; //red
int led_2 = 37; //blue
int led_3 = 44; //green
int led_4 = 47; //yellow

// the setup routine runs once when you press reset:
void setup() {
  // initialize the digital pin as an output.
  pinMode(led_1, OUTPUT);
  pinMode(led_2, OUTPUT);
  pinMode(led_3, OUTPUT);
  pinMode(led_4, OUTPUT);
}

// the loop routine runs over and over again forever:
void loop() {
  digitalWrite(led_4, LOW);
  digitalWrite(led_1, HIGH);
  delay(1000);
  digitalWrite(led_1, LOW);
  digitalWrite(led_2, HIGH);
  delay(1000);
  digitalWrite(led_2, LOW);
  digitalWrite(led_3, HIGH);
  delay(1000);
  digitalWrite(led_3, LOW);
  digitalWrite(led_4, HIGH);
  delay(1000);
}
````

Wenn0101, 2021-03-02:
We let this issue go stale. I am able to control LEDs in the latest release, closing this issue. Comment back if you still have an issue.

closed 2021-03-02 by Wenn0101

beegee-tokyo, 2021-10-02:
@Wenn0101 

I experience the same problem with V2.1.1
I solved it by changing **`Arduino15\packages\SparkFun\hardware\apollo3\2.1.1\variants\SFE_EDGE\config\pins.cpp`** and adding the LED pins in **`variantPinStates[]`**

```c++
/* 
// This file is subject to the terms and conditions defined in
// file 'LICENSE.md', which is part of this source code package.
*/

#include "bridge/pins.h"

const pin_size_t variantPinCount = 8;

PinState variantPinStates[variantPinCount] = {
	{D1, 1, NULL, /*NULL, NULL, NULL,*/ NULL},
	{D38, 38, NULL, /*NULL, NULL, NULL,*/ NULL},
	{D36, 36, NULL, /*NULL, NULL, NULL,*/ NULL},
	{D3, 3, NULL, /*NULL, NULL, NULL,*/ NULL},
	{LED1, 46, NULL, /*NULL, NULL, NULL,*/ NULL},
	{LED2, 37, NULL, /*NULL, NULL, NULL,*/ NULL},
	{LED3, 44, NULL, /*NULL, NULL, NULL,*/ NULL},
	{LED4, 47, NULL, /*NULL, NULL, NULL,*/ NULL},
};
```

beegee-tokyo, 2021-10-02:
@Wenn0101 
And same for BUTTON1. I need to add it to the list, or else the button does not work.

Maybe my code is bad, but I followed the example codes.