sparkfun/Arduino_Apollo3 #306: Confusing repository name?

jerabaul29, 2020-12-09:
As V2 changed to using the mbed-os, I wonder if having 'Arduino' in the name of the repo may be confusing, as this is not an Arduino framework derivative any longer if I understand right?

Wenn0101, 2020-12-10:
@jerabaul29 This is a bit confusing now, but I believe this repo to be appropriately named.

To clarify the purpose of this repo is arduino support for the apollo3 boards. It is the code in this repo that makes our target available in the arduinoIDE.
I believe the confusion comes from the fact that the core functionality comes from an mbed library. 

Our fork of mbed is located [here](https://github.com/sparkfun/mbed-os-ambiq-apollo3).

Another way of phrasing this is that repo is now a layer that is "on top" of mbed. This existence of this layer makes the boards available for use by users of the arduino IDE.

Let me know if this help clarify and if you think there is still an issue that needs addressing.

jerabaul29, 2020-12-10:
Ok, many thanks for your answer! :)

First, of course these are my 2 cents (but I still think I am not the only one who will be confused :) ).

My feeling is that the "Arduino" word is used for 2 things: the "Arduino framework", and the "Arduino IDE". While these were maybe "equal" at the beginning, this is not so true anymore. The "Arduino framework" is available from the "platformio IDE" for example, and now you are pushing some boards code based on the "mbed-OS framework" into the "Arduino IDE". I was wondering therefore if this may be confusing. I know of several embedded software engineers by trade (I do a bit of embedded myself but it is not my trade) who automatically think "Arduino = Arduino framework", i.e. for them the most important when thinking about Arduino is the framework not the IDE, ie Arduino for them means no RTOS and one single-threaded program. Having an mbed-os derived core labeled as Arduino will create some mis-understandings with them, as I suppose this means that one will actually have an RTOS under one's feet, with the full power of the threading / scheduling / etc, right? :)

Wenn0101, 2020-12-29:
I am going to close this issue for now. We aren't taking immediate action on this, but we are open to updating our naming and documentation to reduce confusion. I will be sure to keep this in mind as we continue working on this repo.

closed 2020-12-29 by Wenn0101