sparkfun/Arduino_Apollo3 #309: permission denied uploading on linux - maybe need udev rules?

jerabaul29, 2020-12-11:
Hi,

I get some denied permission errors when trying to upload successfully compiled sketches to an Apollo Red board on Linux (Ubuntu 20.04). I wonder if there may be some need to updates some udev rules? Any idea on this?

Thanks :)

jerabaul29, 2020-12-11:
(sorry, I think the true name is Artemis red board; anyways, this one:

https://www.sparkfun.com/products/15444

jerabaul29, 2020-12-11:
I tried to fix the access, but still not working. At the moment:

```
~$ ls -l /dev/ttyUSB0 
crw-rw---- 1 root dialout 188, 0 des.  11 13:35 /dev/ttyUSB0
~$ whoami
jrmet
~$ groups jrmet
jrmet : jrmet adm dialout cdrom sudo dip plugdev lpadmin lxd sambashare
```

So I would guess writing access should be ok (?), but I still get the error:

```
Sketch uses 118300 bytes (12%) of program storage space. Maximum is 983040 bytes.
Global variables use 27712 bytes (7%) of dynamic memory, leaving 365504 bytes for local variables. Maximum is 393216 bytes.
java.io.IOException: Cannot run program "/home/jrmet/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/tools/uploaders/asb/dist/linux/asb": error=13, Permission denied
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2055)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=13, Permission denied
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

jerabaul29, 2020-12-11:
Ok, after some more debugging, the problem was that the program in charge of uploading the firmware did not have execution rights. I did:

```
~/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/tools/uploaders/asb/dist/linux$ chmod +x asb
```

and now it seems to be able to (try to) upload.

jerabaul29, 2020-12-11:
Do you think there is a way to update the rights on the asb utility installed by the board manager in linux? :)

oclyke, 2020-12-14:
@jerabaul29 thanks for reporting this. we will correct the file permissions. please respond here if you find any more related issues.

@Wenn0101 we should investigate what happens to file permissions in and between these steps:
- generation of executables from the respective uploader repositories ([ASB](https://github.com/sparkfun/Apollo3_Uploader_ASB), [SVL](https://github.com/sparkfun/Apollo3_Uploader_SVL))
- inclusion of executables via submodules in the arduino core
- compression / archival of the arduino core as it is published to a release

jerabaul29, 2021-02-02:
I updated to 2.0.5 today. The issue is still here (it even came back when the non-executable uploader replaced the old uploader that I had made executable).

Nothing bad for me to fix, just a small ```chmod +x```, but for less experienced users this may be a hazzle.

oclyke, 2021-02-15:
hit the same issue on fresh install of v2.0.5

closed 2021-03-22 by Wenn0101