sparkfun/Arduino_Apollo3 #310: cannot upload on board

jerabaul29, 2020-12-11:
On Ubuntu 20.04, after fixing the ```chmod +x``` of #309 , I bump into a new issue: the upload fails. Detailed log:

```
Sketch uses 114812 bytes (11%) of program storage space. Maximum is 983040 bytes.
Global variables use 27320 bytes (6%) of dynamic memory, leaving 365896 bytes for local variables. Maximum is 393216 bytes.
/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/tools/uploaders/asb/dist/linux/asb --bin /tmp/arduino_build_223201/Serial.ino.bin --load-address-blob 0x20000 --magic-num 0xCB -o /tmp/arduino_build_223201/Serial.ino --version 0x0 --load-address-wired 0xC000 -i 6 --options 0x1 -b 115200 -port /dev/ttyUSB0 -r 2 -v 
Header Size =  0x80
original app_size  0x1cd18 ( 118040 )
load_address  0xc000 ( 49152 )
app_size  0x1cd18 ( 118040 )
w0 = 0xcb01cd98
Security Value  0x10
w2 =  0x10008080
addrWord =  0xc000
versionKeyWord =  0x0
child0/feature =  0xffffffff
child1 =  0xffffffff
crc =   0x8834377a
Writing to file  /tmp/arduino_build_223201/Serial.ino_OTA_blob.bin
testing: /tmp/arduino_build_223201/Serial.ino_OTA_blob.bin
Header Size =  0x60
app_size  0x1cd98 ( 118168 )
Writing to file  /tmp/arduino_build_223201/Serial.ino_Wired_OTA_blob.bin
Image from  0x0  to  0x1cd98  will be loaded at 0x20000
Connecting over serial port /dev/ttyUSB0...
Sending Hello.
No response for command 0x00000000
received bytes 1
['0xfc']
Failed to respond
Fail
Sending Hello.
No response for command 0x00000000
received bytes 1
['0xfc']
Failed to respond
Fail
Sending Hello.
No response for command 0x00000000
received bytes 1
['0xfe']
Failed to respond
Fail
Tries = 3
Upload failed
```

Any idea where the problem comes from and how to fix it? I am using v2.0.3, "Red board Artemis", "Ambiq Secure Bootloader", I have tried several baud rates without success, and I am quite very sure that the USB port in use is the right one.

jerabaul29, 2020-12-11:
Ok, got it to work, the 1 value of baudrate I had not tested worked. I had to choose "Sparkfun variable loader boadrate: 230400" and then it worked.

Any idea why this exact value was needed / should this maybe be part of the documentation?

jerabaul29, 2020-12-11:
Mmmh, this gets really confusing. On the next sketch I upload, I need to use "57600". Any idea why the baudrate change from upload to upload / how to set it or know which value to expect? It is quite time consuming to try one after the other until being lucky...

jerabaul29, 2020-12-11:
Mmmh, on the next upload I had to plug in and out altogether and use 57600. Quite confusing.

jerabaul29, 2020-12-11:
Now I do not manage to upload anything any longer.

jerabaul29, 2020-12-11:
And now I manage, after trying randomly different baudrates several times in rows. Any idea about the logics behind?

jerabaul29, 2020-12-12:
Probably related to: https://forum.sparkfun.com/viewtopic.php?f=171&t=53923&p=221527#p221527 .

At this location people mention using another bootloader or something like this to fix the issues; any feedback on that? :)

jerabaul29, 2020-12-12:
Is it possible that using this way of uploading instead may help?

https://github.com/sparkfun/Artemis-Firmware-Upload-GUI

Wenn0101, 2020-12-17:
@jerabaul29 
Can you do a couple steps for me and let me know how it goes? 

1. Connect your board as usual
2.  Select Tools -> Programmer -> Ambiq Secure Bootloader
3.  Select Tools -> Burn Bootloader.
4.  Verify the new bootloader successfully been burned on the device by reading the logs, if not repeat step 3.
5.  Change uploader to SVL. Tools -> Uploader -> SparkFun Variable Loader (Out of Order)
6.  Change Baud to 921600.

Try uploading a few times and let me know your success rate. Note: Do NOT switch to the Ambiq Secure Bootloader (ASB) in between trials, this will blow away the SparkFun Variable Loader (SVL).

When we released the new version of the core, we thought the ASB would be sufficient for most users. Through more testing we have seen that users have had a lot of problems with this bootloader (Especially if you are leaving the arduino COM window open when attempting to upload). In v2.0.3, I pushed a change that should allow the SVL to be used successfully, however the bootloader has to be burned using the aforementioned ASB, once this succeeds we would expect a much better experience using exclusively the SVL.

jerabaul29, 2020-12-17:
Sounds good, many thanks for the explanations! I am away from my board just now, will try later on today. Many thanks for taking the time to look at this problem. I really hope this helps, because at the moment the board is not very usable. Crossing fingers :)

jerabaul29, 2020-12-17:
I think I was able to upload the bootloader (that took quite a few tries):

```
/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/tools/uploaders/asb/dist/linux/asb --bin /home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/tools/uploaders/svl/bootloader/gcc/artemis_module/bin/svl.bin --load-address-blob 0x20000 --magic-num 0xCB -o /home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/tools/uploaders/svl/bootloader/gcc/artemis_module/bin/intermediate --version 0x0 --load-address-wired 0xC000 -i 6 --options 0x1 -b 115200 -port /dev/ttyUSB0 -r 2 -v 
Header Size =  0x80
original app_size  0x328c ( 12940 )
load_address  0xc000 ( 49152 )
app_size  0x328c ( 12940 )
w0 = 0xcb00330c
Security Value  0x10
w2 =  0x10008080
addrWord =  0xc000
versionKeyWord =  0x0
child0/feature =  0xffffffff
child1 =  0xffffffff
crc =   0xc703da31
Writing to file  /home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/tools/uploaders/svl/bootloader/gcc/artemis_module/bin/intermediate_OTA_blob.bin
testing: /home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/tools/uploaders/svl/bootloader/gcc/artemis_module/bin/intermediate_OTA_blob.bin
Header Size =  0x60
app_size  0x330c ( 13068 )
Writing to file  /home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/tools/uploaders/svl/bootloader/gcc/artemis_module/bin/intermediate_Wired_OTA_blob.bin
Image from  0x0  to  0x330c  will be loaded at 0x20000
Connecting over serial port /dev/ttyUSB0...
Sending Hello.
Received response for Hello
Bootloader connected
Received Status
length = 0x58
version = 0x5
Max Storage = 0x4ffa0
Status = 0x2
State = 0x7
AMInfo = 
0x1
0xff2da3ff
0x55fff
0x1
0x4cd00005
0xffffffff
0xffffffff
0xffffffff
0xffffffff
0xffffffff
0xffffffff
0xffffffff
0xffffffff
0xffffffff
0xffffffff
0xffffffff
Sending OTA Descriptor = 0xfe000
Sending Update Command.
number of updates needed = 1
Sending block of size 0x336c from 0x0 to 0x336c
Sending Data Packet of length 8180
Sending Data Packet of length 4984
Sending Reset Command.
Tries = 0
Upload complete!
```

But then I never managed to upload the sketch (tried at least 20 times):

```
/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/tools/uploaders/svl/dist/linux/svl /dev/ttyUSB0 -f /tmp/arduino_build_160478/ThreadsMyTests.ino.bin -b 921600 -v 

Artemis SVL Bootloader
Script version 1.7

Phase:	Setup
	Cleared startup blip
Failed to enter bootload phase

Phase:	Setup
	Cleared startup blip
Failed to enter bootload phase

Phase:	Setup
	Cleared startup blip
Failed to enter bootload phase
Target failed to enter bootload mode. Verify the right COM port is selected and that your board has the SVL bootloader.
```

jerabaul29, 2020-12-17:
Rectification: I was never able to upload with baudrate 921600 in at least 20 tries. I just tried with baudrate 115200, that worked without any issues 5 times in a row. A bit slow, but at least it works :) . Tried a few times 460800, seems to work too.

Wenn0101, 2020-12-17:
Good to hear you were able to get it uploading! Nice call reducing the baud rate until it worked.

jerabaul29, 2020-12-17:
I was quite happy when I saw I could get reliable updates :) . Any idea why the Ambiq bootloader is problematic, and why the baudrate 921600 is problematic?

But the important is that it seems to work fine now. Maybe some documentation / guideline updates would be useful? Do you know if this is reproducible across users? :) .

jerabaul29, 2020-12-19:
I can confirm that this works reliably: I have been playing around all day without issue uploading.

closed 2020-12-29 by Wenn0101